JsBergbau/MiTemperature2 #2: Bouncing temperature compensation

JsBergbau, 2020-01-08:
This is an issue with the LYWSD03MMC Sensor. On successive readings/receiving of the temperature the value bounces a lot. Therefore the debouncing switch has been implemented. It works that way: 
(We skip the rounding which is only used after startup until the second decimal place is not anymore within 0.2 and 0.7). 
Say temperature read is 11.81 degrees. Than it is rounded downwards (floor) to 11.8 If the next reading is 11.86 it is still floored to 11.8
If it reaches 11.7 it rounded up to 11.8
If the next reading is 11.83 it is still rounded up to 11.9
When the next reading is 11.82 it is floored to 11.8
When next reading is 11.85 or 11.86 it is still florred to 11.8

**Due to the way debouncing works it can only be used together with --round option. If round option is not specified, but --debounce than debouncing is ignored.**

I have multiple BME280 and also DS18B20 sensors. Without any measures they show a lot less bouncing values. They just give a lot more decimal places and I'm using only 2 of them. Giving quite stable values.

LYWSD03MMC gives 17 bounces in that measurement without debouncing measures within about 20 minutes
[image]

LYWSD03MMC with debouncing. Within roughly more than half an hour "only" 6 temperature bounces. The humidity is bouncing, but there is no possibility to debounce without losing precision. Things like moving average can still be down when displaying the data 
[image]

--> Debouncing ok

Allen57, 2020-08-19:
Hello, thanks a lot for all your work on this sensor so far. I've recently been toying with it and I wanted to share my own version of "debouncing" temperature and humidity in case it can be useful to someone.

Based on a given debouncing delay (120s for example), I send the temperature and humidity every 120s but I update each of the value, for temperature and humidity, only if they did not "bounced" back to their old value in the 120s delay.

I've made a dumb implementation through a callback but this could be integrated into the original script as well.

<details>
  <summary>[...] --callback debounceSendToDomoticz.sh</summary>

```sh
#!/bin/sh
cd "$(dirname "$0")"

DEBOUNCE_DELAY=120
TMP_DIR=tmp_data_$2
if [ ! -d $TMP_DIR ]; then 
        mkdir -p $TMP_DIR
        touch $TMP_DIR/old_temp
        echo 0 > $TMP_DIR/old_temp_date
        touch $TMP_DIR/old_hum
        echo 0 > $TMP_DIR/old_hum_date
        echo 0 > $TMP_DIR/last_send_date
fi

val_temp=$3
val_hum=$4
val_date=$7

old_temp=$(cat $TMP_DIR/old_temp)
if [ "$old_temp" = "$val_temp" ]; then
        echo "Reset temp date"
        echo $val_date > $TMP_DIR/old_temp_date
fi
old_temp_date=$(cat $TMP_DIR/old_temp_date)
if [ $(($val_date - $old_temp_date)) -ge $DEBOUNCE_DELAY ]; then
        echo "Updating temp value"
        old_temp=$val_temp
        echo $val_temp > $TMP_DIR/old_temp
        echo $val_date > $TMP_DIR/old_temp_date
fi

old_hum=$(cat $TMP_DIR/old_hum)
if [ "$old_hum" = "$val_hum" ]; then
        echo "Reset hum date"
        echo $val_date > $TMP_DIR/old_hum_date
fi
old_hum_date=$(cat $TMP_DIR/old_hum_date)
if [ $(($val_date - $old_hum_date)) -ge $DEBOUNCE_DELAY ]; then
        echo "Updating hum value"
        old_hum=$val_hum
        echo $val_hum > $TMP_DIR/old_hum
        echo $val_date > $TMP_DIR/old_hum_date
fi

last_send_date=$(cat $TMP_DIR/last_send_date)
if [ $(($val_date - $last_send_date)) -ge $DEBOUNCE_DELAY ]; then
        echo $val_date > $TMP_DIR/last_send_date
        echo "sending $old_temp/$old_hum"
        ./sendToDomoticz.sh $1 $2 $old_temp $old_hum $5 $6
fi
```

</details>

The advantage for me is that I can keep the two decimal precisions and this work as well with the humidity.

It can still be improved at the bouncing detection part which only checks if we go back to the exact old value: we could also detect if we go from a higher value to a lower value (or the reverse).

(On a side note, calling a python script every 6 seconds, per sensor, was cpu intensive for my rpi so I switched it to a shell script)

JsBergbau, 2020-08-20:
Hi Allen57,

thanks for sharing. Suppose we have 20.12 °C, so when after 120 seconds it still has 20.12 °C you don't update the value. If it has 20.13 °C you update the new value, correct?
Sounds interesting. What are your experiences? How good does ist work? I mean it could happen that 20.13 °C is reported back, but then temperature bounces again to 20.12 °C, when I have understood your script.

Allen57, 2020-08-20:
To be more accurate, if the first value is 20.12 °C, it sends it immediately. Then the next values, every 6 sec, could be something like 20.13 20.13 20.14 20.12 ... (and after 120s) 20,13. During this delay of 120s, no values are sent, but at the end of the delay, it will send updated values only if the values did not go back to their original value (20.12) during the 120s delay.
In this case, even if the last value is 20.13, it will report 20.12 as this value was reported during the 120s delay.

So if the values are, for example, constantly bouncing back between 20.12 and 20.13, this script will only send the same first value until it stops reporting the first value for more than 120s.

About my experience, I will be able to give you actual data comparison (with and without my script)  in a few hours ;).

Allen57, 2020-08-20:
I've started a test with one sensor using debounce and another one not using it, but then I realized I could do my test based on the same value for the same sensor. So here are the current results (domoticz only keeps one sample per 5 minutes):

<details>
  <summary>Result</summary>

<table>
<tbody>
<thead>
<tr>
<th>With debouncing</th>
<th>Without debouncing</th>
</tr>
</thead>
<tr>
<td>

2020-08-20 14:20:00 | 68 | 25.47
2020-08-20 14:25:00 | 68 | 25.47
2020-08-20 14:30:00 | 68 | 25.47
2020-08-20 14:35:00 | 68 | 25.53
2020-08-20 14:40:00 | 68 | 25.53
2020-08-20 14:45:00 | 68 | 25.53
2020-08-20 14:50:00 | 68 | 25.58
2020-08-20 14:55:00 | 67 | 25.58
2020-08-20 15:00:00 | 67 | 25.64
2020-08-20 15:05:00 | 67 | 25.64
2020-08-20 15:10:00 | 67 | 25.65
2020-08-20 15:15:00 | 67 | 25.65
2020-08-20 15:20:00 | 67 | 25.68
2020-08-20 15:25:00 | 67 | 25.69
2020-08-20 15:30:00 | 67 | 25.74
2020-08-20 15:35:00 | 67 | 25.79
2020-08-20 15:40:00 | 67 | 25.79
2020-08-20 15:45:00 | 67 | 25.82
2020-08-20 15:50:00 | 67 | 25.82
2020-08-20 15:55:00 | 67 | 25.85
2020-08-20 16:00:00 | 66 | 25.85
2020-08-20 16:05:00 | 66 | 25.85

</td>
<td>

2020-08-20 14:20:00 | 68 | 25.45
2020-08-20 14:25:00 | 68 | 25.53
2020-08-20 14:30:00 | 68 | 25.51
2020-08-20 14:35:00 | 68 | 25.53
2020-08-20 14:40:00 | 68 | 25.55
2020-08-20 14:45:00 | 67 | 25.55
2020-08-20 14:50:00 | 67 | 25.6
2020-08-20 14:55:00 | 67 | 25.6
2020-08-20 15:00:00 | 67 | 25.67
2020-08-20 15:05:00 | 67 | 25.65
2020-08-20 15:10:00 | 67 | 25.68
2020-08-20 15:15:00 | 67 | 25.72
2020-08-20 15:20:00 | 67 | 25.71
2020-08-20 15:25:00 | 67 | 25.71
2020-08-20 15:30:00 | 67 | 25.75
2020-08-20 15:35:00 | 67 | 25.8
2020-08-20 15:40:00 | 67 | 25.81
2020-08-20 15:45:00 | 67 | 25.81
2020-08-20 15:50:00 | 67 | 25.85
2020-08-20 15:55:00 | 67 | 25.84
2020-08-20 16:00:00 | 66 | 25.82
2020-08-20 16:05:00 | 66 | 25.83

</td>
</tr>
</tbody>
</table>

</details>