JsBergbau/MiTemperature2 #3: Limit number of measures

FunkyFab, 2020-01-10:
Hi, 

Could it be possible to add a new parameter (--count for instance).
This parameter would be an integer, and the script would exit after receiving this number of measures.
That would allow a limited bluetooth connection.
For instance, I would trigger the script from cron every 5 minutes, with 3 measures. The bluetooth connection would then be maybe 30 seconds every 5 minutes.
Thanks for considering my request.

JsBergbau, 2020-01-10:
Ok I'll implement. Probably I'll implement tomorrow.

FunkyFab, 2020-01-10:
I'm no expert in BLE either, but it seems to me that, to maintain a connection open, you need to send some data on a regular basis.
I have 3 Xiaomi Mija. What I'll do is to keep the connection open on one of them, and poll on the 2 others, and I'll see which strategy is the best, on a battery performance point of view.

JsBergbau, 2020-01-11:
Function was implemented. Please try if it works for you and then close the issue. 

Please report the battery levels after some time, so we know if it saves power. Thx.

FunkyFab, 2020-01-13:
Great! Working perfectly for the moment.
I wrote a callback for sending the info to MQTT.
My home automation software is Jeedom, running on a PI4. Every 5 minutes, it connects to the Zero PI through SSH and starts your Python script.
I now have to find a way to integrate your script with the other option (continuous loop).

JsBergbau, 2020-01-15:
> I now have to find a way to integrate your script with the other option (continuous loop).

Please use the latest version. Due to a bug in bluepy script stopped sometimes collecting data. Now a workaround is implemented.

Which trouble do you have with the continious loop?

FunkyFab, 2020-01-15:
> Which trouble do you have with the continuous loop?
I don't need a measure every 6 seconds. One every 5 minutes is largely enough. My concern is that, by sending data through BTLE every 6 seconds, the battery is drained faster than it would be with a single measure sent every 5 minutes.
I've implemented the integration on my home automation: when the home automation remains more that 5 minutes without any update from the thermometer, it triggers a shell script running your python. For 2 thermometers, it stops after 5 measures, and for the last one, it's looping forever.

wesselah, 2020-01-21:
> Great! Working perfectly for the moment.
> I wrote a callback for sending the info to MQTT.
> My home automation software is Jeedom, running on a PI4. Every 5 minutes, it connects to the Zero PI through [...]

@Funkylab could you share your callback mqtt script i would need an example so i can adapt it to my needs and i am no programming star :-)

FunkyFab, 2020-01-22:
> 
> @Funkylab could you share your callback mqtt script i would need an example so i can adapt it to my needs and i am no programming star :-)

Sure.
Here it is.
[pushToMQTT.zip](https://github.com/JsBergbau/MiTemperature2/files/4095707/pushToMQTT.zip)

wesselah, 2020-01-22:
top

From: FunkyFab
Sent: Wednesday, 22 January 2020 08:08
To: JsBergbau/MiTemperature2
Cc: wesselah; Comment
Subject: Re: [JsBergbau/MiTemperature2] Limit number of measures (#3)

@Funkylab could you share your callback mqtt script i would need an example so i can adapt it to my needs and i am no programming star :-)
Sure.
Here it is.
pushToMQTT.zip

JsBergbau, 2020-01-22:
@wesselah 
THere is also a "sendToMQTT" script in the main repository, provided by @Chiunownow

wesselah, 2020-01-22:
Just did see this also so again thx, it is already working on three sensors.
Found out that a raspberry zero w has the best Bluetooth reach

From: JsBergbau
Sent: Wednesday, 22 January 2020 11:46
To: JsBergbau/MiTemperature2
Cc: wesselah; Mention
Subject: Re: [JsBergbau/MiTemperature2] Limit number of measures (#3)

@wesselah
THere is also a "sendToMQTT" script in the main repository, provided by @Chiunownow

JsBergbau, 2020-01-22:
> ound out that a raspberry zero w has the best Bluetooth reach
Which devices did you test? The Pi Zero W actually is amazing. It goes through 2 thick outer stonewalls and still can read a sensor ~20m away.

wesselah, 2020-01-22:
Raspberry 3 and 4 and both where worse then raspberry zero w which was amazing

From: JsBergbau
Sent: Wednesday, 22 January 2020 13:00
To: JsBergbau/MiTemperature2
Cc: wesselah; Mention
Subject: Re: [JsBergbau/MiTemperature2] Limit number of measures (#3)

ound out that a raspberry zero w has the best Bluetooth reach
Which devices did you test? The Pi Zero W actually is amazing. It goes through 2 thick outer stonewalls and still can read a sensor ~20m away.

giuseppegimo, 2020-01-26:
Hi. First of all thank you for your work.

I'm testing it and I'm experiencing and odd behavior with an "on-demand" activation.

Let's say I want to call it every 10 minutes to get an update on my MQTT broker.

If I try with "-c 1" I don't get the MQTT update:

```
pi@raspberrypi:~/aqaratempsensors $ python3 LYWSD03MMC.py -d XX:XX:XX:XX:XX:XX --round --callback sendToMQTT -c 1
Trying to connect to XX:XX:XX:XX:XX:XX
Temperature: 20.3
Humidity: 54
1 measurements collected. Exiting now.
```

So I try with "-c 2" and I can see the update, but without the battery info (so the script write to the mqtt broker the timestamp as the battery value):

```
pi@raspberrypi:~/aqaratempsensors $ python3 LYWSD03MMC.py -d XX:XX:XX:XX:XX:XX --round --callback sendToMQTT -c 2
Trying to connect to XX:XX:XX:XX:XX:XX
Temperature: 20.3
Humidity: 54

/home/pi/aqaratempsensors/sendToMQTT sensorname,temperature,humidity,timestamp XX:XX:XX:XX:XX:XX 20.3 54 1580030000
Temperature: 20.3
Humidity: 54
2 measurements collected. Exiting now.
```

Then I think to set "-b 1" to retrieve the battery info, and effectively it does, but the battery info sent to the callback script is different:

```
pi@raspberrypi:~/aqaratempsensors $ python3 LYWSD03MMC.py -d XX:XX:XX:XX:XX:XX --round --callback sendToMQTT -c 2 -b 1
Trying to connect to XX:XX:XX:XX:XX:XX
Temperature: 20.4
Humidity: 54
Battery-Level: 99

/home/pi/aqaratempsensors/sendToMQTT sensorname,temperature,humidity,batteryLevel,timestamp XX:XX:XX:XX:XX:XX 20.4 54 0 1580030074
Temperature: 20.3
Humidity: 54
Battery-Level: 99
2 measurements collected. Exiting now.
```

How can I do it?

Thank you.

JsBergbau, 2020-01-26:
Hi giuseppegimo,

thanks for exactly describing your problem. I have understood. I use a thread to send the data to callback. However the main script exits before this thread can send data. I'll fix in the following days. 
I think thats the same with the battery level. The first battery level sent is always 0 % and then the correct batterylevel isn't sent anymore.

JsBergbau, 2020-01-26:
@giuseppegimo 
Was much easier than I thought. I've just updated the script. Please try again. Note first battery readout is always "0".

giuseppegimo, 2020-01-26:
Thank you @JsBergbau , it's ok now.
To avoid the first value equal to 0, I could run the command with "-c 2" and discard the first reading in the callback script.

I understand that the first battery readout is 0, it just seems strange to me that there is the correct value (99) in the output but then it is passed to callback 0:

```
python3 LYWSD03MMC.py -d XX:XX:XX:XX:XX:XX --round -deb --callback sendToMQTT_json -c 2 -b 10
Trying to connect to XX:XX:XX:XX:XX:XX
Temperature: 21.6
Humidity: 52
Battery-Level: 99

/home/pi/aqaratempsensors/sendToMQTT_json sensorname,temperature,humidity,batteryLevel,timestamp XX:XX:XX:XX:XX:XX 21.6 52 0 1580047651
Temperature: 21.6
Humidity: 52
2 measurements collected. Exiting in a moment.
/home/pi/aqaratempsensors/sendToMQTT_json sensorname,temperature,humidity,batteryLevel,timestamp XX:XX:XX:XX:XX:XX 21.6 52 99 1580047657
```

JsBergbau, 2020-01-26:
> I understand that the first battery readout is 0, it just seems strange to me that there is the correct value (99) in the output but then it is passed to callback 0:

Let me explain you:
```
Temperature: 21.6
Humidity: 52
```
Is printed in the callback function receiving the temperature values. This function also sets the measurement data in this case 0 for batterylevel, since we haven't read yet.
Right after that in the mainloop the battery-level is readout, printed in a global variable. This value from the global variable is now used in the callback function to set the batterylevel in the measurement data for all following measurements until the new battery level is readout.

Reading the batterylevel uses extra energy from the sensor, thats why it is implemented that way. If you don't want to stay constantly connected to the aqara, I'd read only temperature and humidity, and once a day or so, I'd read the battery level. The aqara I'm running for more than 3 weeks now, still shows 99% batterylevel.

PS: The debouncing function only works if you run the script constantly. With limiting the number of measurements to a small number it makes no sense.

JsBergbau, 2020-01-27:
@giuseppegimo 
Thanks for the hint with the batterylevel. I've made some thoughts and it was very easy to change the order for reading the batterylevel. 
It should now work to get the correct batterylevel with -c 1. But depeding of the frequency polling the sensor it is mostly not recommended to poll the batterylife for every measurement.

Can you please check if it works for you know as intended?

giuseppegimo, 2020-01-27:
Hi @JsBergbau and thank you.
I think there is still something odd. If I ask for one measure (-c 1), I get two MQTT messages (of which the first with battery=0 and the second one ok):

```
pi@raspberrypi:~/aqaratempsensors $ python3 LYWSD03MMC.py -d XX:XX:XX:XX:XX:XX -b 1 -c 1 --round --callback sendToMQTT_json
Trying to connect to XX:XX:XX:XX:XX:XX
Temperature: 20.4
Humidity: 55
Battery-Level: 99
/home/pi/aqaratempsensors/sendToMQTT_json sensorname,temperature,humidity,batteryLevel,timestamp XX:XX:XX:XX:XX:XX 20.4 55 0 1580158463
Temperature: 20.4
Humidity: 55
1 measurements collected. Exiting in a moment.
/home/pi/aqaratempsensors/sendToMQTT_json sensorname,temperature,humidity,batteryLevel,timestamp XX:XX:XX:XX:XX:XX 20.4 55 99 1580158469
```

Maybe the first callback has to be skipped.

JsBergbau, 2020-01-27:
Hi @giuseppegimo 
Are you sure to use the latest version? The latest version prints the batterylevel before the temperature and humdity output.

Just checked again:

```
./LYWSD03MMC.py -d AA:BB:CC:DD:EE:FF -2p -p2 75 -o2 -4 -p1 33 -o1 -6 -b 1000 --skipidentical 50 -r --debounce --name testroom --callback sendToInflux.sh -c 1
Trying to connect to AA:BB:CC:DD:EE:FF
Battery-Level: 99
Temperature: 8.8
Humidity: 79
Calibrated humidity: 75
1 measurements collected. Exiting in a moment.
/home/pi/XiaomiBluetoothThermometer/sendToInflux.sh sensorname,temperature,humidity,humidityCalibrated,batteryLevel,timestamp testroom 8.8 79 75 99 1580164173
HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: xxxx
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.7.9
X-Request-Id: xxxx
Date: Mon, 27 Jan 2020 22:29:34 GMT
```

giuseppegimo, 2020-01-27:
Yeah there was something wrong, it seems ok now:

```
pi@raspberrypi:~/aqaratempsensors $ python3 LYWSD03MMC.py -d XX:XX:XX:XX:XX:XX -b 1 -c 1 --round --callback sendToMQTT_json
Trying to connect to XX:XX:XX:XX:XX:XX
Connection lost
Waiting...
Trying to connect to XX:XX:XX:XX:XX:XX
Battery-Level: 99
Temperature: 20.3
Humidity: 57
1 measurements collected. Exiting in a moment.
/home/pi/aqaratempsensors/sendToMQTT_json sensorname,temperature,humidity,batteryLevel,timestamp XX:XX:XX:XX:XX:XX 20.3 57 99 1580164997
```

Thanks!

svennd, 2020-01-28:
I'm no expert either; But from what I read, actually staying connected and not pulling data so fast would be better; Cause when you are not connected the sensor is in "advertising mode" so it sends out data packages every so often. If you remain connected and only request data once a minute you only "keep the connection" alive, reducing the overall packages send.

https://stackoverflow.com/questions/36725225/how-can-i-estimate-the-power-consumption-of-a-ble-module?rq=1

Thanks for this useful tool !

JsBergbau, 2020-04-03:
Seems to be solved --> Closing

closed 2020-04-03 by JsBergbau