JsBergbau/MiTemperature2 #5: Python 3.5.3 updated LYWSD03MMC.py

XReyRobert, 2020-01-16:
Hello,

As proposed I would be very grateful of a bug free and updated version of the script for 3.5.3 !
As I'm running Stretch and would like to use callbacks scripts to update remote values...

Many thanks for this work,

JsBergbau, 2020-01-17:
Hi XReyRobert,
if you want to use the callback functions you need Python 3.7. Supporting Python 3.5 would need some bigger rewrite of the script. 

I suggest:
a) Compiling Python 3.7 your own as sugggested in the readme

> If you like installing/compiling Python3.7 please take a look at this tutorial https://gist.github.com/SeppPenner/6a5a30ebc8f79936fa136c524417761d However it took about 5 hours to compile/run the regressiontests on a Raspberry PI3B. I use this compiled version directly without install. If you do, too, you have to change the first line in the script, pointing to your compiled Python version. For bluepy you can copy the bluepy-folder from home/pi/.local/lib/python3.7/site-packages/bluepy to Python-3.7.4/Lib and do a chmod +x bluepy-helper in Python-3.7.4/Lib/bluepy

b) I can provide you a compressed version of the files, which you have to unpack and set the necessary rights. 

What do you prefer?

XReyRobert, 2020-01-18:
Thank you, I went ahead and quickly hacked the 3.5.3 script for my needs.

XReyRobert, 2020-01-18:
A quick share back, the script publish data to virtual temperature and humidity sensors to a Vera home controller via the HTTP api.

https://gist.github.com/XReyRobert/d76c564b6aa65aee85b315f8d2cefb94

closed 2020-01-18 by XReyRobert