JsBergbau/MiTemperature2 #7: Possible to use multiple sensors?

XReyRobert, 2020-01-24:
Hello,

Can you run multiple of this scripts in parallels if you have multiple sensors ? (I don't have yet multiple sensors, so I can't test at the moment...)

JsBergbau, 2020-01-26:
Yeah of course you can run multiple instances of the script. The script was designed that there is no interference at all.
Currently I'm collecting 5 sensors on a Pi Zero W. 

I don't know the hardware limit in the number of devices.  But there seems some, because with 5 sensors running `sudo hcitool lescan --duplicate` often exits. However thats no problem of the script, just a hardware limitation.

giuseppegimo, 2020-01-26:
I can confirm that. I'm using 3 instances running at the same time on a Pi 3.

closed 2020-01-31 by JsBergbau