JsBergbau/MiTemperature2 #9: LYWSD02MMC support?

skibbipl, 2020-01-29:
I successfully connected two LYWSD03MMC sensors but I have also 02 sensor. Can this script support those?

JsBergbau, 2020-01-30:
Sorry, I don't understand. Do you mean a second sensor or another sensor type? If it is another sensor type, than it isn't supported by this script.

skibbipl, 2020-01-30:
Yes, I mean another sensor type. It has e-ink display and clock. But I guess I'll have to stick to this:
https://github.com/h4/lywsd02

jaggil, 2020-02-26:
> Yes, I mean another sensor type. It has e-ink display and clock. But I guess I'll have to stick to this:
> https://github.com/h4/lywsd02

JsBergbau, 2020-04-03:
@skibbipl 
Did you try? Does this script work with LYWSD02 ?

closed 2020-05-17 by JsBergbau

BLahr, 2020-05-18:
Guten Tag,

vielen Dank für Ihre Nachricht.
Ich arbeite nicht mehr für die WKG Software GmbH.
 
Bitte beachten Sie, dass Ihre Email nicht weitergeleitet wird. 
 
Bitte wenden Sie sich je nach Fall an eine dieser beiden Adressen:
support@salesfrontapp.com
telephonesupport@wkg-software.de 

Mit freundlichen Grüßen  

-----------------------------------------------------------------------------------  

Bastian Lahr (lahr@wkg-software.de)  

   

W.K.G Software GmbH  

Hindenburgstr. 38, 30175 Hannover, Germany  

Geschäftsführer: Dipl.-Kfm. Jochen Wunder  

HRB 60250 Amtsgericht Hannover, USt.-ID: DE 813 392 760  

-----------------------------------------------------------------------------------