JsBergbau/MiTemperature2 #10: Measurements interval

fracz, 2020-01-30:
After reading the power-saving discussion in https://github.com/JsBergbau/MiTemperature2/issues/3 I think it would be awesome if we could specify the interval between reading measures while staying connected to the sensor. Something like:

```
./LYWSD03MMC.py --interval 300 ...
```

Would read new data every 5 minutes (and execute the callback, if given).

What do you think?

JsBergbau, 2020-01-31:
There is no "real" measurement interval. As soon as the Xiaomi is once connected with the official app (or this script) it sends out the current readings every ~6 seconds.  Don't be worried about the battery life tto much. My first sensor is now almost one month sending data and has still 99 %.

fracz, 2020-02-01:
Ok, understand.

Premature optimization is root of all evil? ;-)

Thank you!

jaggil, 2020-02-17:
> There is no "real" measurement interval. As soon as the Xiaomi is once connected with the official app (or this script) it sends out the current readings every ~6 seconds. Don't be worried about the battery life tto much. My first sensor is now almost one month sending data and has still 99 %.

I am also interested in knowing how the battery is discharged when the sensor sends notifications every 6 seconds.
Please, could you see my comment in issues # 1 and confirm the battery status with the bytes that are read in the UUID ebe0ccc1-

JsBergbau, 2020-02-24:
I've done some measurements by putting a multimeter between the battery and the Aqara device.
When it is on advertising mode it consumes about 20µA. It is hard to get an exact value because it fluctuates heavily even with a 4000 µF capacitor on the Xiaomi device side. With about 20 µA and a battery capacity of about 210 mAh we get a calculated lifetime of 10500 hours or 437 days with is roughly about 1 year and 2 month.

When connected the values are quite stable and I get around 500 µA power draw and very stable. But I'm a bit confused because 500 µA with a 210 mAh battery gives a batterylifetime of 17,5 days. Either some batteries have more capacity or the multimeter is bad. However some devices failed after quite short time. So round about are the values ok. 
There is no noticable difference in the 500 µA power draw whether the temp/humidity/voltage notification is activated or not. Solely the bluetooth connection uses that much amount of power.

Thats why I'll try to implement a connection intervall of 5 minutes, as described here https://github.com/JsBergbau/MiTemperature2/issues/1#issuecomment-590013238

closed 2020-02-25 by JsBergbau