JsBergbau/MiTemperature2 #12: Script issue

maresob, 2020-02-05:
Hi, 
I have a problem can anyone help??

  File "./LYWSD03MMC.py", line 20
    temperature: float
               ^
SyntaxError: invalid syntax

JsBergbau, 2020-02-06:
Please execute 
`python3 --version`
Looks like you're not using Python 3.7 as required.

CurikovMF, 2020-02-08:
приветствую 
у меня та же проблема 

 File "./LYWSD03MMC.py", line 20
    temperature: float
               ^
SyntaxError: invalid syntax
pi@homebridge:~/MiTemperature2 $ python --version
Python 2.7.13
pi@homebridge:~/MiTemperature2 $ python3.7 --version
Python 3.7.0
pi@homebridge:~/MiTemperature2 $ python3 --version
Python 3.7.0

JsBergbau, 2020-02-10:
Please go to directory where LYWSD03MMC.py and try `python3 LYWSD03MMC.py`

JsBergbau, 2020-04-03:
Seems to be solved --> Closing

closed 2020-04-03 by JsBergbau