JsBergbau/MiTemperature2 #13: Troubleshooting tools and tips

svendickman, 2020-02-10:
I have difficulty getting data from my LYWSD03MMC.   I'm trying to connect from PiZeroW with python 3.7. 
 I can get the MAC with hcitool without any problem.  Your script seems to connect and then loses the connection and then stays in waiting.
  I have tried troubleshooting with gatttool and seem to be able to connect but cannot get any data.   I'm not sure if it is important but I cannot seem to find the meter with the standard Xiaomi phone app, but I can find it with the MijiaTemp application: https://play.google.com/store/apps/details?id=com.smrtprjcts.mijiabt
Any troubleshooting tips would be greatly appreciated.
Thanks in advance.

JsBergbau, 2020-02-10:
>  I'm not sure if it is important but I cannot seem to find the meter with the standard Xiaomi phone app, but I can find it with the MijiaTemp application
Not really important, but try changing the Region to China, then this should work.

How far is the distance between PiZeroW and the Xiaomi?
Try `sudo apt update && sudo apt upgrade`
Then please reboot and try again.

Luilutin, 2020-02-11:
With RPi4 I had similar problem. Moving RPi to closer sensor started work. It is easier scan sensor to get data from it.

JsBergbau, 2020-04-03:
Seems to be solved --> Closing

closed 2020-04-03 by JsBergbau