JsBergbau/MiTemperature2 #14: Cannot connect

jpasantos, 2020-02-12:
Hi, I'm trying your script on a home assistant docker container running **Python 3.7.6**

I have two sensors which I can find:
```
/config/www/scripts # hcitool lescan | grep LYWSD03MMC
A4:C1:38:B7:D7:7F LYWSD03MMC
A4:C1:38:B5:A3:32 LYWSD03MMC
```

But when connecting with `python3 LYWSD03MMC.py  -d A4:C1:38:B5:A3:32` all I get is
```
Trying to connect to A4:C1:38:B5:A3:32
Connection lost
Waiting...
Trying to connect to A4:C1:38:B5:A3:32
Connection lost
Waiting...
Trying to connect to A4:C1:38:B5:A3:32
Connection lost
Waiting...
```

I tried to add some exception details and this is what I have
```
Failed to connect to peripheral A4:C1:38:B7:D7:7F, addr type: public
Traceback (most recent call last):
  File "LYWSD03MMC.py", line 267, in <module>
    p=connect()
  File "LYWSD03MMC.py", line 182, in connect
    p = btle.Peripheral(adress)
  File "/usr/local/lib/python3.7/site-packages/bluepy/btle.py", line 391, in __init__
    self._connect(deviceAddr, addrType, iface)
  File "/usr/local/lib/python3.7/site-packages/bluepy/btle.py", line 439, in _connect
    raise BTLEDisconnectError("Failed to connect to peripheral %s, addr type: %s" % (addr, addrType), rsp)
bluepy.btle.BTLEDisconnectError: Failed to connect to peripheral A4:C1:38:B7:D7:7F, addr type: public
```

Thanks

Luilutin, 2020-02-12:
Have you tried script with shorter distance to the adapter?

jpasantos, 2020-02-12:
> Have you tried script with shorter distance to the adapter?
Hi,
I've put the sensors right next to the bluetooth adapter for all the steps I've wrote in the original issue

JsBergbau, 2020-02-12:
> I've put the sensors right next to the bluetooth adapter for all the steps I've wrote in the original issue

Can you try it with about  30 cm distance? Some devices have problems if they're too near.

If above doesn't work: Can you try to connect with gatttool?

```
gatttool -I
connect AA:BB:CC:DD:EE:FF
#enable notifications
char-write-req 0x0038 0100
#Read battery-Level, consider: note value is in Hex format
char-read-hnd 0x001b
```
If this doesn't work, then I suspect something related with the bluetooth adapter. Or docker, however I'm not a docker expert.

jpasantos, 2020-02-12:
I've tried close and mid distances to no success

I've got this from the command
```
/config # gatttool -I
[                 ][LE]> connect A4:C1:38:B5:A3:32
Attempting to connect to A4:C1:38:B5:A3:32
Error: connect: Connection refused (111)
[A4:C1:38:B5:A3:32][LE]> connect A4:C1:38:B5:A3:32
Attempting to connect to A4:C1:38:B5:A3:32
```

JsBergbau, 2020-02-13:
So there is generally something wrong, independently from the script. 
I would try another BLE device and try to connect to. If it doesn't work, probably something is wrong with your Bluetooth  adapter. If it works probably something is wrong with you Aqara.

jpasantos, 2020-02-13:
Does this work with the raspberry pi4? maybe I should migrate

JsBergbau, 2020-02-14:
Yeah it works with the PI4. However the Bluetooth range of the PI4 was in my tests quite limited. Best results regarding range gives RPI Zero W.

jaggil, 2020-02-17:
> Can you try it with about 30 cm distance? Some devices have problems if they're too near.
>
> If above doesn't work: Can you try to connect with gatttool?
>
> ```
> gatttool -I
> connect AA:BB:CC:DD:EE:FF [...]

Another point to keep in mind is that the sensor only supports one connection. Check if you have, for example, activate / connect an app on your smartphone with which you are monitoring the T / H values

jpasantos, 2020-02-17:
> > > I've put the sensors right next to the bluetooth adapter for all the steps I've wrote in the original issue
> > 
> > 
> > Can you try it with about 30 cm distance? Some devices have problems if they're too near.
> > If above doesn't work: Can you try to connect with gatttool?
> > ```
> > gatttool -I
> > connect AA:BB:CC:DD:EE:FF
> > #enable notifications
> > char-write-req 0x0038 0100
> > #Read battery-Level, consider: note value is in Hex format
> > char-read-hnd 0x001b
> > ```
> > 
> > 
> > If this doesn't work, then I suspect something related with the bluetooth adapter. Or docker, however I'm not a docker expert.
> 
> Another point to keep in mind is that the sensor only supports one connection. Check if you have, for example, activate / connect an app on your smartphone with which you are monitoring the T / H values. This has happened to me, but I'm using ESP32

I haven't connected to the phone since it only supports the china server and I don't want to use that

JsBergbau, 2020-02-18:
Theoretically a neighbour could also connect to your sensor. To check that no one is connected watch the display. No bluetooth sign may appear. As  soon it appears somebody is connected.

jaggil, 2020-02-18:
You can test the communication with the following android app:
https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp

The app also allows you to see the data of each UUID

jpasantos, 2020-02-20:
I setup a PI 2 with raspbian and using the same adapter I managed to get it to work. So it should be something on the docker container screwing it up

JsBergbau, 2020-04-03:
Seems to be solved --> Closing

closed 2020-04-03 by JsBergbau

akostadinov, 2022-01-22:
For anybody else hitting this, make sure you don't have a typo in MAC address.