JsBergbau/MiTemperature2 #17: [Feature Request] Connection lost callback

jorisvaesen, 2020-02-21:
Would it be possible to implement a connection lost callback? This would make it easier to debug the reliability of the connection or make a trigger when the connection is lost for more than x minutes.

JsBergbau, 2020-02-23:
Yeah that would be possible, But is it worth it? Whats your standard callback? If it is a database or a MQTT you just could query when was the last value reported.

jorisvaesen, 2020-02-27:
My standard callback is just a curl to a custom backend i've build. Checking the last sync would make it less accurate since I use `-skip 50` which makes it more of a guess if the connection is lost when it's lost for a short time. It would be useful to log all connection lost error through the callback to see how many times the connection is actually lost and trying some options to reduce it in the future like replacing the Pi or the sensor.

JsBergbau, 2020-02-27:
If you just want to know how often connection is lost, you could just redirect the output to a log file. Then you grep for "Connection lost" and you know how often connection gets lost. Thats already implemented and works instantly.

closed 2020-03-31 by jorisvaesen