JsBergbau/MiTemperature2 #18: Powerdraw with this script 20 times higher than with official app

JsBergbau, 2020-02-24:
I've done some measurements by putting a multimeter between the battery and the Aqara device.
When it is on advertising mode it consumes about 20µA. It is hard to get an exact value because it fluctuates heavily even with a 4000 µF capacitor on the Xiaomi device side. With about 20 µA and a battery capacity of about 210 mAh we get a calculated lifetime of 10500 hours or 437 days with is roughly about 1 year and 2 month.

When connected the values are quite stable and I get around 500 µA power draw and very stable. But I'm a bit confused because 500 µA with a 210 mAh battery gives a batterylifetime of 17,5 days. Either some batteries have more capacity or the multimeter is bad. However some devices failed after quite short time. So round about are the values ok. 
There is no noticable difference in the 500 µA power draw whether the temp/humidity/voltage notification is activated or not. Solely the bluetooth connection uses that much amount of power.

Thats why I'll try to implement a connection intervall of 5 minutes, as described here https://github.com/JsBergbau/MiTemperature2/issues/1#issuecomment-590013238

Update: With this simple multimeter there can be seen no difference in power draw when doing an active scan with `sudo hcitool lescan --duplicate`. On active scanning mode the device replies to the scans with its device name.

And now the very interessting part: When you use the official Xiaomi Mi App the power draw goes to about 500-700 µA shortly and then it falls down to about 25 µA even is the sensor is still connected with the app. If you touch and hold it so that the temperature rises it is almost immediately shown on smartphone. If we could reach that small power draw we could stay connected because there is only a very small additional energy usage and establishing the connection uses most energy.

So the question is whats going on that with official app the power draw is that kind of small and with this script so insane high.

michaelnimbs, 2020-02-25:
Wasn't there some encrypted advertisement payload data only the official app could decrypt?

Could you check the power consumption with this unofficial app? [https://play.google.com/store/apps/details?id=com.smrtprjcts.mijiabt](url)
(Would be easier to reverse engineer / decompile I guess)

JsBergbau, 2020-02-25:
Yeah, I've tested with that App. It only connects for a few seconds, gets the values and then disconnects. While that the powerdraw is between 500-700 µA like the official app. There is no longer period when the App stays connected and I couldn't find any option to let the App stay connected.

JsBergbau, 2020-02-25:
I've also tried with this App https://play.google.com/store/apps/details?id=com.freetimeprojects.mijiareader but it also only shortly connects.
 
Then I connected with nrf connect https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp It had about 400 µA. Later I wanted to find out why 100 µA lower, but then it also had about 500 µA. Don't know why. Was there really a difference? Or is it a problem with the multimeter used?

JsBergbau, 2020-02-25:
I've created a bluetooth hci log and the app wrote a lot of stuff to the sensor. I'll try to send and see whats happening to the power consumption.

michaelnimbs, 2020-02-25:
Thank you for your tremendous effort!

jaggil, 2020-02-25:
@JsBergbau 
Hi, I just wrote the value 0xF40100 in the following characteristic and the consumption drops tremendously.
  set conn interval -> "ebe0ccd8-"
Please, can you try it on your devices?

JsBergbau, 2020-02-26:
@jaggil 
Thank you for all your support with the Aqara Bluetooth Thermometer. I think our answers have just overlapped. I'm now also writing `char-write-req 0x0046 f40100` and the power consumption drops. It is now implemented in the official MiTemperature2 script.

JsBergbau, 2020-04-03:
Since it is fixed for a while, I close now.

closed 2020-04-03 by JsBergbau

eranl, 2021-02-01:
I just had a first battery die after two weeks and another after one week. My RP4 is within 30cm of the sensor. Using stock firmware and a constant connection. Interestingly, the first battery in a second sensor which is located outside and has a spotty connection lasted three weeks.
As a workaround, I'll now try a wrapper script that will connect and take one measurement every 5 minutes.