JsBergbau/MiTemperature2 #20: HA component

Monacoslo, 2020-02-27:
Since this is the only known working method to get the data from this new xiaomi sensor we would really appreciate if you can also create Home Assistant component!

JsBergbau, 2020-02-28:
Hi I understand your wish to have it in Home Assistant. However I have no clue about integrating into home assistant. As I'm using Openhab, I know there are a lot of ways to import data. You could use MQTT for  example https://www.home-assistant.io/integrations/mqtt/
There's also a callback in this project you just have to adopt.

JsBergbau, 2020-04-03:
Seems to be solved --> Closing

closed 2020-04-03 by JsBergbau

AnthonyKNorman, 2020-04-17:
Have a look at my repository. I am successfully using @JsBergbau 's code with Home Assistant.