JsBergbau/MiTemperature2 #22: [Feature Request] Adapter selection option

wiewiur, 2020-03-28:
I think possibility of selecting bluetooth adapter if we have more than one (hci0 hci1 etc.) would be very useful.

JsBergbau, 2020-03-29:
Hi,

could you describe your usecase for multiple interfaces a bit more? I'm just curious.

I've implemented interface selection, use -i or --interface 1 for hci1 and so on.

Please give me short feedback if it works the way you wanted.

wiewiur, 2020-03-30:
I have raspbery PI and I wanted to use external BT dongle beacuse of range issues with internal BT module. Im testing and it looks OK for now.
Thanks

JsBergbau, 2020-03-30:
What PI do you have? In the last days there was a firmware issue with reduced bluetooth range. It should be fixed now. If you still have problems with range you can try `sudo apt-get install bluez=5.50-1+rpt1`

wiewiur, 2020-03-30:
RPI4 in metal case :)

JsBergbau, 2020-03-31:
The Armorcase https://www.berrybase.de/neu/armor-geh-228-use-f-252-r-raspberry-pi-4-schwarz or another? With that the antenna stays free, it just gets a bit more directional :)

wiewiur, 2020-03-31:
No, its full case: https://www.aliexpress.com/item/4000494307148.html

JsBergbau, 2020-03-31:
Ok, you really should use an external adapter :)

closed 2020-03-31 by JsBergbau