JsBergbau/MiTemperature2 #26: when executed via cron no data is sent to influxdb via callback

mjazwiecki, 2020-04-06:
Perhaps not an issue with script itself (great job, btw!) but when run on cron, it does not send data to influx via callback. once run from terminal, works fine. 

measureSensor01Now.sh
```
#!/bin/bash
./LYWSD03MMC.py -d A4:C1:38:87:25:CF -b -r --debounce --skipidentical 50 --name sensor01 -c 10 --callback sendToInflux.sh
```
cron:
```
*/15 * * * * /home/mat/xiaomiSensorToInflux/measureSensor01Now.sh >/dev/null 2>&1
*/15 * * * * /home/mat/xiaomiSensorToInflux/measureSensor02Now.sh >/dev/null 2>&1
```
grep CRON /var/log/syslog:
```
Apr  6 19:30:01 middleMan CRON[28597]: (mat) CMD (/home/mat/xiaomiSensorToInflux/measureSensor02Now.sh >/dev/null 2>&1)
Apr  6 19:30:01 middleMan CRON[28599]: (mat) CMD (/home/mat/xiaomiSensorToInflux/measureSensor01Now.sh >/dev/null 2>&1)
```
any help would be much appreciated!
thanks

blackdot, 2020-04-07:
Try in your "measureSensor01Now.sh" script use full path for LYWSD03MMC.py.
Something like:
`#!/bin/bash
/home/mjazwiecki/LYWSD03MMC.py -d A4:C1:38:87:25:CF -b -r --debounce --skipidentical 50 --name sensor01 -c 10 --callback sendToInflux.sh`

mjazwiecki, 2020-04-07:
yes! ah silly me 😄

closed 2020-04-07 by mjazwiecki