JsBergbau/MiTemperature2 #28: Measured values red don't change

tuccler, 2020-04-14:
Hello there,
i'm using this repository since 2 or 3 releases and i really like it. But now i have some problems and i can't figure out why. 

I'm using the Mitemperature2 v1.8 script three times in seperate instances. After some time (~2 hours), the script will allways output the same values even if the temperature/humidity changes. 

I noticed that the Raspberry Pi Zero W reacts really slowly while connected via SSH. The connection via bluetooth to the LYWSD03MMCs is maintained.

My start values are:
./LYWSD03MMC.py -d "MAC" -r -b

JsBergbau, 2020-04-14:
Can you please execute `apt list --installed | grep bluez` and post the result here?

tuccler, 2020-04-15:
This is the output of the command:
bluez-firmware/testing,now 1.2-4+rpt2 all [installed,automatic]
bluez/now 5.50-1+rpt1 armhf [installed,upgradable to: 5.50-1.2~deb10u1+rpt1]

JsBergbau, 2020-04-15:
Looks good so far. I have the same versions.
I had a quite similar problem. I've rebooted the Wifi-AP and then all worked flawlessly again. I would also try to cut the power of AP and RPI Zero for about 30 seconds (you can shutdown the PI before).

JsBergbau, 2020-04-16:
So when the script outputs always the same values, even if temperature changes, what happens if you restart the script? Are then the correct values displayed again?

tuccler, 2020-04-20:
If i restart the script, the first Output is fine. Every output after that won't change. I cloned the project again from github and now it's working fine. I will keep an eye on that.

closed 2020-05-17 by JsBergbau