JsBergbau/MiTemperature2 #29: Nice works -- a question

DazWilkin, 2020-04-15:
Thank you for this.

Your code works for me.

I've been trying to get a Golang example working but I don't understand the mapping between the devices handles and characteristics.

Your code appears to use `0x0038` and `0x0046`

How were you able to identify these handles?

If I enumerate my devices handles:

```
handle = 0x0038, uuid = 00002902-0000-1000-8000-00805f9b34fb
handle = 0x0046, uuid = ebe0ccd8-7a0a-4b0c-8a1a-6ff2997da3a6
```
I'm unable to find characteristics `00002902*`

But I do have:

```
Service: ebe0ccb07a0a4b0c8a1a6ff2997da3a6
  Characteristic  ebe0ccd87a0a4b0c8a1a6ff2997da3a6
    properties    write 
  Descriptor      2901 (Characteristic User Description)
    value         73657420636f6e6e20696e74657276616c00 | "set conn interval\x00"
```

Am I understanding the mapping correctly?

Thanks!

JsBergbau, 2020-04-15:
Hi. You're welcome.

Finding the handle corresponding the the characteristics was really not easy. Since I was completly new to BLE it was a lot of try until it finally worked. I used nRF-Connect App for Android and especially for finding the handle for energy saving I used a hci-log from Android. And as far as I remember there was directly the handle used and no characteristics. 

Hope I could help you a bit.

DazWilkin, 2020-04-15:
That's very helpful! Thank you

closed 2020-04-15 by DazWilkin