JsBergbau/MiTemperature2 #30: Connection lost => RPi temp. higher and higher

MASZer, 2020-04-22:
Hello,
I am newbie in general RPi and coding, I have RPi from few days.

I use this script to collecting data from my 4 xiaomi sensors.
Everythig work well, I use crontab to collect 1 data every 15 minutes from each sensor, but...
When I close door in my badroom sometimes I lost connecion with sensor there (I have RPi 4 with not good bluetooth range).
I noticed that then RPi ty to reconnect and roconnect, my CPU use more and more resources and temperature  rise and rise (over 70 degrees). Reboot help for short time...

Could anyone help me to modify "my code" to achive: when connection lost stop/interrupt script?

"my code" in crontab:

`/home/pi/mi-temp/LYWSD03MMC.py -d A4:C1:38:6C:65:46 -r -b -c 1 --callback sendToDomoticz14.py
/home/pi/mi-temp/LYWSD03MMC.py -d A4:C1:38:80:56:FE -r -b -c 1 --callback sendToDomoticz15.py
/home/pi/mi-temp/LYWSD03MMC.py -d A4:C1:38:40:76:76 -r -b -c 1 --callback sendToDomoticz16.py
/home/pi/mi-temp/LYWSD03MMC.py -d A4:C1:38:46:6C:78 -r -b -c 1 --callback sendToDomoticz17.py`

Thanks in advance!

JsBergbau, 2020-04-23:
Hi.

Problem is the bad connection.
The script tries to connect and exits only if it got one measurement. Because of the weak signal it seems this is often not possible after 15 minutes.
So after 15 minutes the next instance of the script starts. Possibly blocking each other. To prevent, just run one script

```
#!/bin/bash

while true
do
    /home/pi/mi-temp/LYWSD03MMC.py -d A4:C1:38:6C:65:46 -r -b -c 1 --callback sendToDomoticz14.py >dev/null
    sleep 15m
done
``` 
You can start that script with @reboot in crontab and to do without reboot with a `nohup <scriptname> /dev/null &`

MASZer, 2020-04-23:
Hello,
thank you for answer. As I wrote I am newbie in RPi and coding.

The code should looks like which option?:

1)

while true
do
    /home/pi/mi-temp/LYWSD03MMC.py -d A4:C1:38:6C:65:46 -r -b -c 1
--callback sendToDomoticz14.py >dev/null
    sleep 15m
    /home/pi/mi-temp/LYWSD03MMC.py -d A4:C1:38:80:56:FE -r -b -c 1
--callback sendToDomoticz15.py >dev/null

done

MASZer, 2020-04-23:
Hello,
thank you for answer.
As I wrote, I am totally newbie in RPi and coding.

How my code should finally looks like that?

*n crontab:*

@reboot
while true
do
    /home/pi/mi-temp/LYWSD03MMC.py -d A4:C1:38:6C:65:46 -r -b -c 1
--callback sendToDomoticz14.py >dev/null
    sleep 15m
   /home/pi/mi-temp/LYWSD03MMC.py -d A4:C1:38:80:56:FE -r -b -c 1
--callback sendToDomoticz15.py >dev/null
   sleep 15m

done

Best regards

closed 2020-04-25 by MASZer