JsBergbau/MiTemperature2 #34: Problem with Domoticz

almostlunatic, 2020-07-06:
Hi,

I can't get a virtual sensor on Domoticz to be updated. I changed the IP-address to "127.0.0.1" and set the "idx_temp" value.

With a command:

`sudo python3 LYWSD03MMC.py -d A4:C1:38:C3:A0:01 -r -deb -b -c 2  --callback sendToDomoticz.py`

I get the response:

`/home/riho/mijia2/sendToDomoticz.py sensorname,temperature,humidity,voltage,batteryLevel,timestamp A4:C1:38:C3:A0:01 27.7 48 2.666 56 1594067106
Temperature: 27.7
Humidity: 48
Battery voltage: 2.666
Battery level: 56
2 measurements collected. Exiting in a moment.
`

But the values of the sensor on Domoticz don't update. What could be wrong?

JsBergbau, 2020-07-07:
Sorry, as I have no experience with Domoticz I can't help you here. I suggest asking at the Domoticz communitiy for help.

almostlunatic, 2020-07-09:
Somehow, the script started to work. But then I met another problem. One of my 3 sensors stopped to talk with the script. The script tries to connect and stays this way eternally. The 2 others work well, one just worked one day and this is it. It is listed on BT Manager on the Domoticz machine, it could be recognized by the apps Mi Home, MijiaTemp, and Mija Reader, but it doesn't respond to the script. I removed the battery, tried to reset it - nothing helps.  I'm out of ideas, why the script doesn't connect it...

JsBergbau, 2020-07-09:
Try reducing the distance between sensor and reading BT device.

almostlunatic, 2020-07-10:
I tried it, I kept the sensor just some centimeters from the BT device, now it's 2 meters, while the 2 others that work are on the next floor. Doesn't help.

martsevoymo, 2020-09-16:
good day.  Help.  I have a username and password to log into domoticz.  tell me how to properly edit the script so that domoticz receives data from the temperature and humidity sensor?

Su4oNoK, 2020-10-09:
> good day. Help. I have a username and password to log into domoticz. tell me how to properly edit the script so that domoticz receives data from the temperature and humidity sensor?

HI to use username and password to log into domoticz, i'm edit script "sendToDomoticz.py" - Add domoticzrequest and remove res = requests.get. Now it work

**Code**

_#!/usr/bin/python3
import os
import re
import requests
import time
import sys
**import urllib.request
import base64
domoticzserver = "192.168.1.2:8080"**
idx_temp = "26"
**domoticzusername = "Admin"
domoticzpassword = "AdminPassword123"
base64string = base64.encodebytes(('%s:%s' % (domoticzusername, domoticzpassword)).encode()).decode().strip()
def domoticzrequest (url):
 request = urllib.request.Request(url)
 request.add_header("Authorization", "Basic %s" % base64string)
 response = urllib.request.urlopen(request)
 return**
val_temp = sys.argv[3]  # change to sys.argv[5] for calibrated
val_hum = sys.argv[4]
val_bat = sys.argv[6]
val_comfort = "0"
if float(val_hum) < 40:
    val_comfort = "2"
elif float(val_hum) <= 70:
    val_comfort = "1"
elif float(val_hum) > 70:
    val_comfort = "3"
**domoticzrequest("http://" + domoticzserver + "/json.htm?type=command&param=udevice&idx=" + idx_temp + "&nvalue=0&svalue=" + val_temp + ";" + val_hum + ";" + val_comfort +  "&battery=" + val_bat)**_

-----------------------------------------------------------------------------------------------------------------
**to cron i use**

_*/5 * * * * /usr/bin/python3.7 /home/Admin/dev-domoticz/scripts/python/MiTemperature2/LYWSD03MMC.py -d 00:00:00:00:00:00 --count 1 --unreachable-count 5 -2p -p2 75 -o2 -4 -p1 33 -o1 -6 -b 5 --callback sendToDomoticz.py_

guessvic, 2020-11-09:
> HI to use username and password to log into domoticz, i'm edit script "sendToDomoticz.py" - Add domoticzrequest and remove res = requests.get. Now it work
>
> **Code**
>
> _#!/usr/bin/python3 import os [...]

Hello Su4oNoK,

I keep getting the following error when using above codes, the only thing different from mine is that I have opened a file and put the output data when executing ./LYWSD03MMC.py ... into a .txt file. That way I can store data from MiTemperature2 every 6-8 seconds whilst have the data uploaded to Domoticz at a specific time. Everything worked without the Authentication, but pain in the butt when it is required...

Traceback (most recent call last):
  File "/usr/lib/python3.7/ast.py", line 35, in parse
    return compile(source, filename, mode, PyCF_ONLY_AST)
  File "/home/pi/MiTemperature2/sendToDomoticz.py", line 23
    request = urllib.request.Request(url)
          ^
IndentationError: expected an indented block

Here is the modified code:

`#!/usr/bin/python3.7

import os
import re
import requests
import time
import sys
import urllib.request
import base64

file = open("/home/pi/MiTemperature2/xm.txt", "r")
_lst = list(file.readlines())
file.close()
last = _lst[len(_lst)-1]
init = last.find('sensorname,temperature,humidity,voltage,timestamp xm') + 53
_sub = last[init:]
_end = _sub[0:].split(" ")

domoticzserver = "192.168.1.2:8080"
idx_temp = "1"
domoticzusername = "Admin"
domoticzpassword = "AdminPassword123"
base64string = base64.encodebytes(('%s:%s' % (domoticzusername, domoticzpassword)).encode()).decode().strip()

def domoticzrequest (url):
request = urllib.request.Request(url)
request.add_header("Authorization", "Basic %s" % base64string)
response = urllib.request.urlopen(request)
return

val_temp = _end[0]
val_hum = _end[1]
val_bat = _end[2]

val_comfort = "0"
if float(val_hum) < 40:
val_comfort = "2"
elif float(val_hum) <= 70:
val_comfort = "1"
elif float(val_hum) > 70:
val_comfort = "3"

domoticzrequest("http://" + domoticzserver + "/json.htm?type=command&param=udevice&idx=" + idx_temp + "&nvalue=0&svalue=" + val_temp + ";" + val_hum + ";" + val_comfort + "&battery=" + val_bat)
`

Please help me!! Thank you...

Su4oNoK, 2020-11-09:
> > > good day. Help. I have a username and password to log into domoticz. tell me how to properly edit the script so that domoticz receives data from the temperature and humidity sensor?
> > 
> > 
> > HI to use username and password to log into domoticz, i'm edit script "sendToDomoticz.py" - Add domoticzrequest and remove res = requests.get. Now it work
> > **Code**
> > _#!/usr/bin/python3 import os import re import requests import time import sys **import urllib.request import base64 domoticzserver = "192.168.1.2:8080"** idx_temp = "26" **domoticzusername = "Admin" domoticzpassword = "AdminPassword123" base64string = base64.encodebytes(('%s:%s' % (domoticzusername, domoticzpassword)).encode()).decode().strip() def domoticzrequest (url): request = urllib.request.Request(url) request.add_header("Authorization", "Basic %s" % base64string) response = urllib.request.urlopen(request) return** val_temp = sys.argv[3] # change to sys.argv[5] for calibrated val_hum = sys.argv[4] val_bat = sys.argv[6] val_comfort = "0" if float(val_hum) < 40: val_comfort = "2" elif float(val_hum) <= 70: val_comfort = "1" elif float(val_hum) > 70: val_comfort = "3" **domoticzrequest("http://" + domoticzserver + "/json.htm?type=command&param=udevice&idx=" + idx_temp + "&nvalue=0&svalue=" + val_temp + ";" + val_hum + ";" + val_comfort + "&battery=" + val_bat)**_
> > **to cron i use**
> > _*/5 * * * * /usr/bin/python3.7 /home/Admin/dev-domoticz/scripts/python/MiTemperature2/LYWSD03MMC.py -d 00:00:00:00:00:00 --count 1 --unreachable-count 5 -2p -p2 75 -o2 -4 -p1 33 -o1 -6 -b 5 --callback sendToDomoticz.py_
> 
> Hello Su4oNoK,
> 
> I keep getting the following error when using above codes, the only thing different from mine is that I have opened a file and put the output data when executing ./LYWSD03MMC.py ... into a .txt file. That way I can store data from MiTemperature2 every 6-8 seconds whilst have the data uploaded to Domoticz at a specific time. Everything worked without the Authentication, but pain in the butt when it is required...
> 
> ```
> Traceback (most recent call last):
>   File "/usr/lib/python3.7/ast.py", line 35, in parse
>     return compile(source, filename, mode, PyCF_ONLY_AST)
>   File "/home/pi/MiTemperature2/sendToDomoticz.py", line 23
>     request = urllib.request.Request(url)
>           ^
> IndentationError: expected an indented block
> ```
> 
> **Here is the modified code I use:**
> 
> ```
> #!/usr/bin/python3.7
> 
> import os
> import re
> import requests
> import time
> import sys
> import urllib.request
> import base64
> 
> file = open("/home/pi/MiTemperature2/xm.txt", "r")
> _lst = list(file.readlines())
> file.close()
> last = _lst[len(_lst)-1]
> init = last.find('sensorname,temperature,humidity,voltage,timestamp xm') + 53
> _sub = last[init:]
> _end = _sub[0:].split(" ")
> 
> domoticzserver = "192.168.1.2:8080"
> idx_temp = "1"
> domoticzusername = "Admin"
> domoticzpassword = "AdminPassword123"
> base64string = base64.encodebytes(('%s:%s' % (domoticzusername, domoticzpassword)).encode()).decode().strip()
> 
> def domoticzrequest (url):
> request = urllib.request.Request(url)
> request.add_header("Authorization", "Basic %s" % base64string)
> response = urllib.request.urlopen(request)
> return
> 
> val_temp = _end[0]
> val_hum = _end[1]
> val_bat = _end[2]
> 
> val_comfort = "0"
> if float(val_hum) < 40:
> val_comfort = "2"
> elif float(val_hum) <= 70:
> val_comfort = "1"
> elif float(val_hum) > 70:
> val_comfort = "3"
> 
> domoticzrequest("http://" + domoticzserver + "/json.htm?type=command&param=udevice&idx=" + idx_temp + "&nvalue=0&svalue=" + val_temp + ";" + val_hum + ";" + val_comfort + "&battery=" + val_bat)
> ```
> 
> Please help me!! Thank you...

@guessvic HI

most likely problems due to indentation in the file.
I am attaching the original file
https://drive.google.com/file/d/1L6z9xPP5pT3SVpdKMUwyM78oxqA1-88f/view?usp=sharing
mistake
[image]

guessvic, 2020-11-10:
> @guessvic HI
>
> most likely problems due to indentation in the file.
> I am attaching the original file
> https://drive.google.com/file/d/1L6z9xPP5pT3SVpdKMUwyM78oxqA1-88f/view?usp=sharing
> mistake
> [image]

Hello @Su4oNoK ,

Thank you so much!!!

cybermandiant, 2020-12-13:
super excelent thanks

closed 2021-01-03 by JsBergbau