JsBergbau/MiTemperature2 #35: Iterate over multiple devices

grunci, 2020-07-08:
Hi,
Is it possible somehow today to specify multiple devices to iterate over from command line?

Regards,
Peter

JsBergbau, 2020-07-08:
Hi Peter,

no its not possible. One instance polls the values from one device. Changing this would require a major rewrite. So Just exceute the script multiple times if you want to get values from multiple devices.

closed 2020-07-08 by JsBergbau

Ferni7, 2020-07-09:
Grunci may have thought that you can only query one device per bluetooth interface. But you can actually run this script on multiple devices at once with a single BT interface.

grunci, 2020-07-09:
Hi @Ferni7 ,
I managed to create a call which fulfills my requirement.
`cat addresses.txt | awk {'print $1'} | xargs -I {} python3 LYWSD03MMC.py -d {} -c 1 -i 0`

I was just curious if it is possible to invoke the script only once for a couple of sensors. I have 4 at the moment.
If major rewrite is needed, then my solution is good enough.