JsBergbau/MiTemperature2 #36: Retrieving logged data

Juul, 2020-07-23:
According to [this](https://des.gbtcdn.com/uploads/pdm-desc-pic/Electronic/image/2019/12/13/20191213152954_41283.jpg) the sensor stores three months of values locally. 

Does anyone know how to retrieve this logged data?

jaggil, 2020-07-23:
There can be done a lot more with that sensor. It stores highest and lowest values at hour level, has an integrated realtime clock and a few things more like changing the values for the comfort icon on the display. Credits go to jaggil who investigated on this and documented it well. You find the results here https://github.com/JsBergbau/MiTemperature2/issues/1 Just search for jaggil.

JsBergbau, 2020-07-23:
Hi Juul,

have a look at https://github.com/JsBergbau/MiTemperature2/issues/1#issuecomment-591106624

Currently one of my sensors never connected to the script has about 1298 values stored according to 

[image]

Juul, 2020-07-23:
Wow thanks! Exactly what I was looking for. Great work!

guessvic, 2020-09-01:
Hi JsBergbau,

Thank you for all the effort! I am newbie to all this. I want to know how did you know your equipment has stored 1298 values from that value (0x) 11-05-00-00-12-05-0-00?

I've tried your solution connecting LYWSD003MMC with instance to my raspberry pi 4 8gb and send them to domoticz, but I also want to keep records when LYWSD003MMC advertises its results every 6 seconds locally. I've read your instruction on using callback with a bash file so data can be written whilst LYWSD003MMC is triggered, but only one way or the other. So this intrigued me as I saw the device itself can store data upto 3 months locally, but I just don't know how to retrieve them.

Please help... Thanks!

JsBergbau, 2020-09-02:
Hi  guessvic,

> Thank you for all the effort! I am newbie to all this. I want to know how did you know your equipment has stored 1298 values from that value (0x) 11-05-00-00-12-05-0-00?

Because of 12-05-00-00 is in "human byte order" 05-12 so 512 Hex which is 1298 decimal.

> I've tried your solution connecting LYWSD003MMC with instance to my raspberry pi 4 8gb and send them to domoticz, but I also want to keep records when LYWSD003MMC advertises its results every 6 [...]

I've also never tried to retrieve locally stored values, so I can't help. Sorry.

guessvic, 2020-09-02:
> Hi guessvic,
>
>
> Because of 12-05-00-00 is in "human byte order" 05-12 so 512 Hex which is 1298 decimal.
>
>
> I've also never tried to retrieve locally stored values, so I can't help. Sorry.

Thank you for the reply with answers. I will keep up searching and if I find something, I will share with you. Again, big thank you for the efforts put into this MiTemperature2!!!

pvvx, 2021-03-04:
All firmware versions save the measurement history. In addition to the firmware version from ATC1441.
An example of reading history from the original Xiaomi LYWSD03MMC:
https://github.com/pvvx/ATC_MiThermometer/issues/54

qwandor, 2021-03-04:
In case it's useful, this is a utility in Rust to fetch all historical data from a set of sensors and dump it to InfluxDB: https://github.com/alsuren/mijia-homie/blob/master/mijia-homie/src/mijia-history-influx.rs

The actual code to fetch historical data is here in `get_all_history`, if you want to reimplement it: https://github.com/alsuren/mijia-homie/blob/master/mijia/src/lib.rs#L424

JsBergbau, 2021-03-04:
> 
> 
> All firmware versions save the measurement history. In addition to the firmware version from ATC1441.
> An example of reading history from the original Xiaomi LYWSD03MMC:
> [pvvx/ATC_MiThermometer#54](https://github.com/pvvx/ATC_MiThermometer/issues/54)

Thanks. A very helpful tool. Where does the LYWSD03MMC get its time from? I've never set the time on this sensor.

qwandor, 2021-03-05:
There's a characteristic (`ebe0ccb7-7a0a-4b0c-8a1a-6ff2997da3a6`) to get and set it. Unfortunately the onboard clock is not very accurate. Here's a utility to check the time and fix it: https://github.com/alsuren/mijia-homie/blob/master/mijia/examples/fix-clock.rs

Note that historical data stored on the sensor retains the time from when it was saved though, even if that was wrong.