JsBergbau/MiTemperature2 #37: not a real issue... but

poseidogithub, 2020-07-27:
Hello! many thanks for teh script, works like a charm with my domoticz.

one questio from a newbie, what would be the method to have multiple thermomethers? I see no options on the scripts, so I guess the only current way without coding is to have one script per sensor?

thanks!

JsBergbau, 2020-07-27:
Hi poseidogithub,

> so I guess the only current way without coding is to have one script per sensor?

exactly. One instance of the script per sensor. Changing this would be a lot of work and because of hanging bluepy issue currently even not really possible.

poseidogithub, 2020-07-27:
Thanks for the quick reply!

Im thinking.. maybe a 3rd script that calls MiTemperature2 passing the mac of each sensor as a variable, and the IDX of the virtual sensor of domotics also as a value?

here we go... to learn Pyton! ahaha

poseidogithub, 2020-07-27:
well I did a very uglhy thing, but it works:

I have 6 sensors, so I created 6 identical SendToDomoticzX.py (being  X from 1 to 6) each one with the idx of the sensor in domiticz (6 different), then a bash script that calls 6 times /LYWSD03MMC.py  (one with each MAC) and with teh callback pointing to the tight SendToDomoticzX.py

thanks!

closed 2020-07-28 by poseidogithub