JsBergbau/MiTemperature2 #39: option request: stop the script if connection is lost

poseidogithub, 2020-07-28:
Hello, 

first of all, Thanks a lot for the script, works like a Charm!

I've created a script that cycles trough different sensors (connects to sensor number one, gets a reading, sends it to domoticz and disconnects, connects to sensor number 2, gets a reading, sends it to domoticz and disconnects, and so on.

Issue: 

When trying to connect to get a reading there is a very usseful option (--count N) htat allos to get ONE reading and finish the script, but when the connection is not working, (maybe out of batterie, out of range, etc) the script keeps querying the thermometer and never stops, so In my scenario this means that if this happens on sensor2  It will loop trying to get sensor 2 data, and I'll never get sensor 3.

sudo python3 MiTemperature2-3/LYWSD03MMC.py -c 1 -d A4:C1:38:E9:0B:0E -2p -p2 75 -o2 -4 -p1 33 -o1 -6 -b 5 --callback sendToDomoticz.py

Trying to connect to A4:C1:38:E9:0B:0E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E9:0B:0E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E9:0B:0E
Connection lost
Waiting...
Trying to connect to A4:C1:38:E9:0B:0E

is there a away to have a variable as --count N that specifies the number of connection retries instead to loop to infinite?

thanks!

almostlunatic, 2020-07-29:
The same problem for me, I have 3 sensors, each of them tends to time out. Then the scripts hang the interface and have to be killed.

Trashwarez, 2020-07-29:
Same here. Would be really fine to have an option about the number of connection retries

JsBergbau, 2020-07-30:
Hi @all.
Since so many need this feature, it is now implemented. Use `--unreachable-count N` to exit after N unsuccessful connection attempts.

closed 2020-07-30 by JsBergbau

poseidogithub, 2020-07-30:
Great, thanks!!!