JsBergbau/MiTemperature2 #45: problem int

cybermandiant, 2020-11-15:
hello i have a problem with this script 
when I want to send the data to my home automation, the script answers me

pi@raspberrypi:~ $ /usr/bin/python3 /home/pi/domoticz/scripts/python/MiTemperature2/LYWSD03MMC.py -d A4:C1:38:B0:A0:FE -r -deb -b -c 1 -call ./sendToDomoticz.py
Trying to connect to A4:C1:38:B0:A0:FE
Temperature: 20.0
Humidity: 73
Battery voltage: 2.869 V
Battery level: 77
1 measurements collected. Exiting in a moment.
argument of type 'int' is not iterable
Traceback (most recent call last):
  File "/home/pi/domoticz/scripts/python/MiTemperature2/LYWSD03MMC.py", line 78, in thread_SendingData
    if ' ' in mea.sensorname:
TypeError: argument of type 'int' is not iterable

cybermandiant, 2020-11-15:
i have read all issues 
i have python 3.7

JsBergbau, 2020-11-15:
Hi cybermandiant,
thanks for your detailed reported. Allowed me to find the problem very fast. I've rewritten the script to suppprt ATC mode, see https://github.com/JsBergbau/MiTemperature2/issues/44 and there this bug came in.

Please download the latest version and it should work. I've just fixed the bug.

cybermandiant, 2020-11-15:
hello hello i try this and you send my rapport ;)

cybermandiant, 2020-11-15:
excelent thanks for your reactivity is work perfectly ;)
pi@raspberrypi:~/domoticz/scripts/python/MiTemperature2 $ /usr/bin/python3 /home/pi/domoticz/scripts/python/MiTemperature2/LYWSD03MMC.py -d A4:C1:38:B0:A0:FE -r -deb -b -c 1 -call ./sendToDomoticz.py
Trying to connect to A4:C1:38:B0:A0:FE
Connection lost
Waiting...
Trying to connect to A4:C1:38:B0:A0:FE
Temperature: 20.0
Humidity: 73
Battery voltage: 2.878 V
Battery level: 78
1 measurements collected. Exiting in a moment.
/home/pi/domoticz/scripts/python/MiTemperature2/./sendToDomoticz.py sensorname,temperature,humidity,voltage,batteryLevel,timestamp A4:C1:38:B0:A0:FE 20.0 73 2.878 78 1605437716

closed 2020-11-25 by JsBergbau