JsBergbau/MiTemperature2 #47: hello new problem

cybermandiant, 2020-11-28:
i have installed domoticz on new raspi4
python 3.7 installed i have this 
pi@raspberrypi:~ $ /usr/bin/python3 /home/pi/domoticz/scripts/python/MiTemperatu                                                                                                        re2/LYWSD03MMC.py -d A4:C1:38:B0:A0:FE -r -deb -b -c 1 -call ./sendToDomoticz.py
Trying to connect to A4:C1:38:B0:A0:FE
Connection lost
Waiting...
Trying to connect to A4:C1:38:B0:A0:FE
Temperature: 18.7
Humidity: 70
Battery voltage: 2.924 V
Battery level: 82
1 measurements collected. Exiting in a moment.
/home/pi/domoticz/scripts/python/MiTemperature2/./sendToDomoticz.py sensorname,t                                                                                                        emperature,humidity,voltage,batteryLevel,timestamp A4:C1:38:B0:A0:FE 18.7 70 2.9                                                                                                        24 82 1606554984
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 159, in _new                                                                                                        _conn
    (self._dns_host, self.port), self.timeout, **extra_kw)
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 80, in                                                                                                         create_connection
    raise err
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 70, in                                                                                                         create_connection
    sock.connect(sa)
OSError: [Errno 113] No route to host

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 600, in                                                                                                         urlopen
    chunked=chunked)
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 354, in                                                                                                         _make_request
    conn.request(method, url, **httplib_request_kw)
  File "/usr/lib/python3.7/http/client.py", line 1244, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1290, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1239, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/usr/lib/python3.7/http/client.py", line 1026, in _send_output
    self.send(msg)
  File "/usr/lib/python3.7/http/client.py", line 966, in send
    self.connect()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 181, in conn                                                                                                        ect
    conn = self._new_conn()
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 168, in _new                                                                                                        _conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.HTTPConnection object                                                                                                         at 0xb5fd3b50>: Failed to establish a new connection: [Errno 113] No route to h                                                                                                        ost

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/usr/lib/python3/dist-packages/urllib3/connectionpool.py", line 638, in                                                                                                         urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/lib/python3/dist-packages/urllib3/util/retry.py", line 398, in incr                                                                                                        ement
    raise MaxRetryError(_pool, url, error or ResponseError(cause))
urllib3.exceptions.MaxRetryError: HTTPConnectionPool(host='192.168.1.36', port=8                                                                                                        0): Max retries exceeded with url: /json.htm?type=command&param=udevice&idx=52&n                                                                                                        value=0&svalue=18.7;70;1&battery=82 (Caused by NewConnectionError('<urllib3.conn                                                                                                        ection.HTTPConnection object at 0xb5fd3b50>: Failed to establish a new connectio                                                                                                        n: [Errno 113] No route to host'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/domoticz/scripts/python/MiTemperature2/./sendToDomoticz.py", li                                                                                                        ne 36, in <module>
    + val_bat
  File "/usr/lib/python3/dist-packages/requests/api.py", line 75, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/api.py", line 60, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 533, in reque                                                                                                        st
    resp = self.send(prep, **send_kwargs)
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 646, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.1.36', por                                                                                                        t=80): Max retries exceeded with url: /json.htm?type=command&param=udevice&idx=5                                                                                                        2&nvalue=0&svalue=18.7;70;1&battery=82 (Caused by NewConnectionError('<urllib3.c                                                                                                        onnection.HTTPConnection object at 0xb5fd3b50>: Failed to establish a new connec                                                                                                        tion: [Errno 113] No route to host'))
Data couln't be send to Callback, retrying...

JsBergbau, 2020-11-28:
Hi,

the problem is with sending the data to domoticz. There I can't really help, but the error message looks like your domoticz system can't be reached because of a network connectivity error. Did you set the correct IP in sendToDomoticz.py script? 
Since I don't use domoticz my possibilities to help are quite limited.

cybermandiant, 2020-11-28:
yes yes I don't know what's going on anymore none of my domoticz work but homebridge yes .... thank you for the answer

cybermandiant, 2020-11-28:
I found the solution to the problem
the connection port was not good in my installation of domoticz I had put the connection port 8899 despite the modification of the sendtodomoticz file the connection did not want to do this suddenly I went back to 8080 and all work again at the top thank you for all :)

closed 2020-11-29 by JsBergbau