JsBergbau/MiTemperature2 #49: Add license information

vazhnov, 2020-12-08:
Hello!

I found link to this project from https://hackaday.com/2020/12/08/exploring-custom-firmware-on-xiaomi-thermometers/

Could you please add information about license of this repository — is it allowed to copy, change and use the code in any purpose?
It would be great to see any popular open source license.

JsBergbau, 2020-12-08:
Hello vazhnov,

thanks for your inquiry. I'm thinking about a license text. However this could take some time. in the meantime: Do you have any special usecase you want to use the software?

vazhnov, 2020-12-08:
Actually I don't have a plan to use this code.

Usually when a person publish any code, he thoughts "anybody can use it". And forgets about the license. But in most countries, this means the code is proprietary.

My hobby is to remind people to add any opensource license, if they want it.

As I see, you have 54 forks and at least 5 of them have some changes. Without proper license, they are proprietary forks of proprietary product, and this may not have any legal status, so they are legally doomed, I think…

JsBergbau, 2020-12-08:
Licensing is a very complex thing. 
As long as nobody sells this code it is ok to copy, change and use it for private purposes.

If a company wants to sell this code, it depends. For example if a company wants to use this code for some energy management to save energy based on the reported temperature it would be ok for me. But if they just sell the code it would be not ok. 

If a company wants to use the code within their company (not selling) it depends upon company, if I would agree for them to use it. If for example a big company like Facebook with a lot of money wants to use it, I'd rather disagree with free use. If a local small store wants to use this software it would be ok.

I think it is hard to pack such use cases into a standard license. Or is possible? Do you have some experiences with licenses and could you recommend one based upon these thoughts?

vazhnov, 2020-12-08:
If you want to limit commercial usage, then opensource license is definitely not what you want. Any popular opensource license allows **any** usage.
Create your own license — I think it is bad idea. Too many things to know, no proved usage cases. Maybe it is better to mention in README, that you still thinking about the license and add link to this ticket :)

therealbstern, 2021-02-17:
I think you want a Creative Commons license. There are a lot of variations and you can probably find one that likes resale but permits reuse.

https://creativecommons.org/licenses/

I don't think I can help narrow it down further, because only you know what you want to allow.