JsBergbau/MiTemperature2 #51: Permission denied?

Luzr, 2020-12-14:
Not sure if its me or the script but when I run `./LYWSD03MMC.py --atc`  I get a permission denied. Anything I should check? I also ran` sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)`. I'm running on a pi zero w.

error
```
user@raspberrypi:~/MiTemperature2 $ ./LYWSD03MMC.py --atc
-bash: ./LYWSD03MMC.py: Permission denied
```

JsBergbau, 2020-12-15:
Seems like you missed making the script itself executable.
Try `chmod +x LYWSD03MMC.py` and it should be fine.

Luzr, 2020-12-16:
That was it. Thanks for the help.
us@iot:~/MiTemperature2 $ ./LYWSD03MMC.py --atc
Script started in ATC Mode
----------------------------
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
BLE packet: A4:C1:38:xx:xx:xx 00 1111111111111111111111111-59
Temperature:  12.2
Humidity:  37
Battery voltage: 3.076 V
RSSI: -59 dBm

closed 2020-12-21 by Luzr