JsBergbau/MiTemperature2 #52: Python 3.9 compatibility?

skibbipl, 2020-12-16:
I've recently upgraded python to 3.9.1 and I'm getting following error:
```
# /usr/local/bin/LYWSD03MMC.py -a
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a namefile and --namefileonlydevices is specified.
Also --name Argument is ignored, if you require names, please use --namefile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights,
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Traceback (most recent call last):
  File "/usr/local/bin/LYWSD03MMC.py", line 421, in <module>
    toggle_device(dev_id, True)
  File "/usr/local/bin/bluetooth_utils.py", line 122, in toggle_device
    fcntl.ioctl(hci_sock.fileno(),
OSError: [Errno 19] No such device
```

JsBergbau, 2020-12-17:
How do you execute Python? Please exec `head LYWSD03MMC.py` and `which python3` and post the results here. 
I've installed Python3.9 via compiling and had a similiar issue because of other python executable. 

You can also try `python3 LYWSD03MMC.py --atc`
If this works than check the shebang in the first line.

skibbipl, 2020-12-17:
```
# head LYWSD03MMC.py
#!/usr/bin/python3 -u
#!/home/openhabian/Python3/Python-3.7.4/python -u
#-u to unbuffer output. Otherwise when calling with nohup or redirecting output things are printed very lately or would even mixup

from bluepy import btle
import argparse
import os
import re
from dataclasses import dataclass
from collections import deque
```
```
# which python3
/usr/bin/python3
```
```
/usr/bin/python3 -V
Python 3.9.1
```

JsBergbau, 2020-12-17:
Very strange, so granting the permissions worked.

Can you try `hciconfig` ? What is printed there.

skibbipl, 2020-12-17:
Hmm, nothing is displayed after running `hciconfig`. Bluetooth is enabled (this is raspberry pi 4).

JsBergbau, 2020-12-17:
Ok, then that is the problem. No bluetooth device is found. Also the error `OSError: [Errno 19] No such device` says that. However I thought that it could be something with permissions.

Have you already rebooted? RPI 4 is very strange with bluetooth. If I plug an USB bluetooth adapter I have hci0 and hci1. Then after reboot only hci0 is there with the MAC address of USB apdater. So internal adapter is not available when booted with plugged USB adapter.

skibbipl, 2020-12-17:
Yeah, simple reboot solved the issue. It seems something died since my last reboot. Problem solved.

closed 2020-12-17 by skibbipl