JsBergbau/MiTemperature2 #53: Never finds my ATC flashed sensors

jds11111, 2020-12-19:
I have followed your instructions on two different PiS for installation. I run the command with the --atc switch, and it reports no errors, but just shows that it is listening without ever discovering any sensors. I also went through the Troubleshooting reccomendaations, but no go. The sensors picked up and read properly by my ESP32 modules programmed with ESPhome. This failure happened for two different RPi0WS running buster.

Any suggestions to troubleshoot?

JsBergbau, 2020-12-19:
Hi jds11111
Seems something is disabling BLE scanning. Please try using `--watchdogtimer 5` and if still nothing is displayed, please post the full log after 2-3 minutes runtime. BLE scanning is often disabled if other application access bluetooth, espcially when the maintain connections to other devices.

jds11111, 2020-12-19:
Hi JsBergbau,

Thanks for the tip. Where is the log?  I don't see anything in the MiTemperature2 directory. I have had the script running for several minutes (using the new switch) without any detection.  I let it run for pretty long, because I have it set to 5min between advertisements on my sensors, but the script never saw them.

JsBergbau, 2020-12-19:
The log is directly printed on console, just copy that output and post it here.

jds11111, 2020-12-19:
OK, here it is:
```
pi@raspberrypi-cam3:~/Programs/MiTemperature2 $ ./LYWSD03MMC.py --atc --watchdogtimer 5
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a namefile and --namefileonlydevices is specified.
Also --name Argument is ignored, if you require names, please use --namefile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights, 
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
```

JsBergbau, 2020-12-19:
And for minutes nothing more is reported than that?

jds11111, 2020-12-19:
Correct.  Here is the output (which looks problematic)

```
pi@raspberrypi-cam3:~/Programs/MiTemperature2 $ sudo hcitool lescan --duplicate
Set scan parameters failed: Input/output error
```

JsBergbau, 2020-12-19:
Yeah thats the problem. Your PI can't scan. I have this error when connected to about 7 sensors. Then scanning works a few seconds and then stops. 
You can try to reboot and redo the scan command. If it works then, start one application after another to find that one that stops your BLE scanning.

jds11111, 2020-12-19:
I have tried rebooting several times.  It has never detected a device. Not sure what you mean about starting applications. I am running a pretty minimal headless pi.

JsBergbau, 2020-12-19:
I also use headless PIs using Raspian minimal OS. 
Perhaps it is something internal. Did you try to cut power for about 30 seconds? However I only needed once cutting to get full functionality back power after a lighting strike into a house 200 meters away.

Your problem is quite hard to track. If cutting power doesn't help, please begin with a fresh installation and  flash "Raspberry Pi OS Lite" from https://www.raspberrypi.org/software/operating-systems/ install everything needed for running sudo hcitool lescan --duplicate
I'm quite sure this even runs out of the box. If that works you can try the script again.

jds11111, 2020-12-19:
Power cycling did not fix it.  Fresh re-install of the OS, and still the same problem.   Maybe because the pi is an older version?

```
pi@raspberrypi-cam3:~/Programs/MiTemperature2 $ sudo hcitool lescan --duplicate
Set scan parameters failed: Input/output error
```

JsBergbau, 2020-12-19:
`pi@raspberrypi-cam3`
I don't consider this as suitable for tracking down the error. You changed for example the hostname. A fresh install is `pi@raspberrypi` 

Please reboot and execute
`sudo hcidump` for 5 minutes and post the output here. Do not execute anything else after reboot while hcidump is running.
If nothing is accessing your bluetooth you should see only
```
sudo hcidump
HCI sniffer - Bluetooth packet analyzer ver 5.50
device: hci0 snap_len: 1500 filter: 0xffffffff
```
If so running then `./LYWSD03MMC.py --atc --watchdogtimer 5` should work. I use this script on 3 devices, 2 Pi Zero W and one Raspberry 3B and everywhere it runs fine. So its strange that you have the same problem on two Rasbperry PI Zero Ws.

jds11111, 2020-12-19:
Really?  Changing the hostname is problematic?  What about the region, and telling wifi what frequencies to use, etc?

My pi does not recognize the command `hcidump`.

JsBergbau, 2020-12-19:
> Really? Changing the hostname is problematic? What about the region, and telling wifi what frequencies to use, etc?
No these things shouldn't be problematic, but if they would how do you know when you change them before testing? 

> My pi does not recognize the command hcidump

Please install via `sudo apt update && sudo apt install bluez-hcidump`

jds11111, 2020-12-19:
My edit crossed in the ether:  see  above.

JsBergbau, 2020-12-19:
> pi@raspberrypi-cam3:~/Programs/RPi_Cam_Web_Interface

You really installed RPi Cam Web Interface on a "fresh" install? Thats what I meant. You should flash the images as written above and then test my script. If it works, then do step by step your modifcations and between each step test if it still works. RPI Cam Webinterface installs so many packets. It could be that there is some standard behaviour modified for example.

jds11111, 2020-12-19:
No, the problem was there before installing that package. I just changed the hostname for the sake of my router, mDNS, etc.

jds11111, 2020-12-19:
I can do it again, for the sake of debugging.

jds11111, 2020-12-20:
I reisntalled the OS. Upgraded nothing. Installed git to download your script (which should probably be added to the instructions, BTW). Installed the Python stuff suggested. Otherwise nothing is changed. Your script still fails, and see the same behavior. Namely, the script turns off Bluetooth.. I ran rasping-config to set language, locale, region, etc. Reboot and check again. Still fails. Did sudo apt update, and upgrade. Same outcome from your script.

JsBergbau, 2020-12-20:
What image did you use?

jds11111, 2020-12-20:
Lite, 32bit

JsBergbau, 2020-12-20:
Just tried it myself.
Dowloaded 2020-12-02-raspios-buster-armhf-lite.zip and flashed and booted in a Raspberry Pi Zero W. Doing all the steps as described in README.md https://github.com/JsBergbau/MiTemperature2/blob/master/README.md executed 
`pi@raspberrypi:~/MiTemperature2 $ ./LYWSD03MMC.py --atc` and it worked. It showed a lot of my sensors. 
However you're right. After running my script hcitool could not scan again. This is fixed now.

So please post the output `hcitool lescan` after one minute here.

jds11111, 2020-12-20:
Well, I don't want to post publicly all the MAC addresses that it picks up. However, here is what happens. First, to make sure that the BLE is enabled I run `sudo hciconfig hci0 down && sudo hciconfig hci0 up`.  Then, I run `sudo hcitool lescan`.  It sees at least 5 MAC addresses per second.  In about 8 seconds it picks up 5 ATC flashed devices. Then I run `./LYWSD03MMC.py --atc`.  After several minutes it reports nothing.  I kill it and run `sudo hcitool lescan` again:
```
pi@raspberrypi:~/Programs/MiTemperature2 $ ./LYWSD03MMC.py --atc
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a namefile and --namefileonlydevices is specified.
Also --name Argument is ignored, if you require names, please use --namefile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights, 
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
^Cpi@raspberrypi:~/Programs/MiTemperature2 $ sudo hcitool lescan
Set scan parameters failed: Input/output error
``` 
And again:
```
pi@raspberrypi:~/Programs/MiTemperature2 $ sudo hciconfig hci0 down && sudo hciconfig hci0 up
pi@raspberrypi:~/Programs/MiTemperature2 $ sudo hcitool lescan
LE Scan ...
xx:zz:aa:ff:66:44 (unknown)
xx:zz:aa:ee:66:55 (unknown)
xx:yy:bb:dd:66:33 (unknown)
```
Including the ATC devices. 

Does that cover what you wanted to see?

JsBergbau, 2020-12-20:
Please download the latest version which fixed that hcitool lescan fails after exiting the script. While running this is normal, only one instance can scan. If you try to execute hcitool lescan in parallel the second will also fail.

Please always use for testing  `./LYWSD03MMC.py --atc --watchdogtimer 5`
only this ensures that scanning is reenabled if anything else accessed bluetooth and thus scanning was disabled.

It is very strange that lescan only reports unknown. Normally first result is unknown, then second result is the device name.

```
 sudo hcitool lescan
LE Scan ...
XX:XX:XX:XX:XX:XX (unknown)
XX:XX:XX:XX:XX:XX ATC_3AD389
XX:XX:XX:XX:XX:XX (unknown)
XX:XX:XX:XX:XX:XX ATC_7F6C3E
XX:XX:XX:XX:XX:XX (unknown)
XX:XX:XX:XX:XX:XX LYWSD03MMC
XX:XX:XX:XX:XX:XX (unknown)
XX:XX:XX:XX:XX:XX CC-RT-BLE
XX:XX:XX:XX:XX:XX (unknown)
XX:XX:XX:XX:XX:XX MJ_HT_V1
```

So it is very strange that this script doesn't work for you, whereas it works here everywhere and no similiar reports so far.

JsBergbau, 2020-12-20:
When did you flash your devices / download the flashed firmware? Perhaps something in the ATC firmware changed so this script doesn't recognize your ATC flashed devices.

jds11111, 2020-12-20:
I flashed it last week, so it is recent. However, how would that explain the fact that the BLE becomes disabled immediately after running the script?

Re: the observation about unknown devices. There is typically a string of unknown MAC stresses, then a known address labeled 'unknown' immediately followed by the same address labeled ATC112233, as you describe.

My pi was flashed using RPiImager, which automatically grabs the recent image (supposedly).

JsBergbau, 2020-12-20:
> how would that explain the fact that the BLE becomes disabled immediately after running the script?

It is not disabled, it is just used by my script, so hcitool can't use lescanning. Just try it. Open one SSH connection, exec sudo hcitool lescan and then in a second ssh session try simulaneously running sudo hcitool lescan again. You will get the same error. 

> My pi was flashed using RPiImager, which automatically grabs the recent image (supposedly).

I also use Raspberry Pi Imager, however I've downloaded latest version from website and flashed that. 

Can you upload your flashed firmware somewhere, then I take a look at it, but I guess I have the same version flashed.

JsBergbau, 2020-12-20:
Update: How did you configure your sensors? 

[image]

What did you select here? Custom or Mi Like? For this script it has to be "custom" which is the default when you flash the firmware.

jds11111, 2020-12-20:
Good point.  I use Mi Like, because that is what is needed by ESPhome to work. 

I was hoping to use a mix of these already existing devices (ESP chips and RPis) in my house to pick up the sensors, no matter where they are. I also use the Passive BLE integration in home assistant running on my RPi, which reads the Mi Like sensors fine. It sounds like your script would not be compatible with that setup.

jds11111, 2020-12-20:
> It is not disabled, it is just used by my script, so hcitool can't use lescanning. Just try it. Open one SSH connection, exec sudo hcitool lescan and then in a second ssh session try simulaneously [...]

Not sure how to do that. Imager says that it is released 2020-12-02.

jds11111, 2020-12-20:
OK, that was it.  I changed one of the sensors to Custom, and then your script could read it.  Too bad it is [not compatible with ESPhome](https://github.com/atc1441/ATC_MiThermometer#using-custom-firmware-with-esphome).

closed 2020-12-21 by jds11111