JsBergbau/MiTemperature2 #54: BlueGiga support?

curlyel, 2020-12-21:
The LYWSD03MMC.py is built on `bluepy` so it works with `bluez` Linux Bluetooth implementation.

It therefore depends on a `bluez` supported Bluetooth adaptors (obviously) and limits the use to Linux only.

**Mind supporting (alternatively) Bluegiga's BGAPI through `pygatt `too?**
https://www.silabs.com/wireless/bluetooth/bluegiga-low-energy-legacy-modules/device.bled112

This would open the script working on **non-Linux** systems as well allowing to use **"remote" serial Bluetooth dongles** from Bluegiga:

> pygatt provides a Pythonic API by wrapping two different backends:
> 
> - BlueZ (requires Linux), using the gatttool command-line utility.
> - Bluegiga's BGAPI, compatible with USB adapters like the BLED112.

Sorry, if this is a dumb question and the api's are too different to easily move from one to the other ;-)

JsBergbau, 2020-12-21:
Hi curlyel,

pygatt is a wrapper around gatttool. 
In my tests gatttool didn't notice when connection was lost to device

> However it didn't notice when BT connection was lost, while this Python-Script automatically reestablishes the connection.

In addition there are used some linux commands like pstree to workaround a bug in bluepy. 

How far away should your bluetooth dongle be from the device? I used remotely an USB dongle connected over 20m of ethernet cable. SInce USB needs 4 wires, I used 2 wires for data and the rest for power. Worked fine.

curlyel, 2020-12-21:
Thanks for the quick reply 👍 

> gatttool didn't notice when connection was lost to device

Was expecting such technical reasons :-(

> In addition there are used some linux commands

Well - some more dependencies - understood.

> I used remotely an USB dongle connected over 20m of ethernet cable

Yes, this would be an option ;-) I even have some "active" usb-via-Cat.5 transceivers here - which work almost o.k.

BUT: I prefer serial dongles for the zoo of wireless devices/gadgets/... and connect them all via `ser2net/socat` to the home automation controller. Therefore having old RaspberryPi's with read-only filesytem in  good RF-positions holding all the Zigbee/ZWave/RFXCom/**BlueGiga's** and "serving" them via network to the controller.

Was just wondering if I could use your script to collect the measurements from the Xiaomi-thermometers in ATC mode "remotely". Have to rethink how to proceed with the ATC thermometers then ;-)

Again: Thanks for providing the script and answering my question.
Understood the reason and dependencies better now -> Not an easy change . Closing ;-)

closed 2020-12-21 by curlyel

JsBergbau, 2020-12-21:
> Therefore having old RaspberryPi's with read-only filesytem in good RF-positions holding all the Zigbee/ZWave/RFXCom/BlueGiga's and "serving" them via network to the controller.

Have you considered to use USB over network / IP ? There is some software around for that. So on your RaspberryPI with good reception position you install the USB server and on the home automation controller you use the client and get the bluetooth interface there. Here are some software examples listed https://stackoverflow.com/questions/18837277/using-a-remote-hosts-usb-port-as-local-usb-linux-and-windows

But still you can let your RaspberryPI in read only file system. It can receive and transmit the data directly via network. 

If you still want to use it with ATC mode, then we don't need pygatt. In fact if you manage to permanently run an bluetooth LE scan by some command on your receiver, then we can "hack" the script to give you the output there. 

Still I would prefer your central RaspberryPI as receiver and then send the decoded data via network. You can write your custom callback to get the data to your controller in the format it needs.

curlyel, 2020-12-21:
> considered to use USB over network / IP ?
Yes, I'm aware of that. Unfortunately, I already have quite some BLE112 here already (six, to be precise ;-)) and try to put them into operation for this. Even if I connect them via USB-IP, they will still appear as serial devices offering just the BGAPI :-(

> Still I would prefer your central RaspberryPI as receiver and then send the decoded data via network. You can write your custom callback to get the data to your controller in the format it needs

Understand. Would consider this, probably via MQTT which is already an "input channel". But the above (sticking with the serial BLE112) is preventing us from doing so because of lacking BGPAPI support - right?

> run an bluetooth LE scan by some command on your receiver, then we can "hack" the script to give you the output there

Don't how to do LE scan with the BlueGiga's from the command line.
If I manage this, I guess it's not that difficult to extract the measurements since the the broadcasted information is a single hex string containing the values. But this would be out the scope of your script (even though I probably can leverage the decoding part for the ATC from it :-) )