JsBergbau/MiTemperature2 #55: Unable to Send to Influx DB

Luzr, 2020-12-21:
I'm not sure if you can help but I'll give it a try. It's most likely an issue on my setup but some guidance would be great. I used [this ](https://github.com/jkehres/docker-compose-influxdb-grafana)Docker setup and I can "ping" the influx DB from the RaspberyPi but the --data-binary sending isnt working. 

I made sendToInflux.sh executable by chmod -x sendToInflux.sh and I get the error "--data-binary: command not found". This is running on a Raspberry Pi 3.

Here's the error when I try to send to InfluxDB.
```
/home/pi/MiTemperature2/sendToInflux.sh sensorname,temperature,humidity,voltage,humidityCalibrated,timestamp [mac] 20.4 38 3.065 0 1608569300
/home/pi/MiTemperature2/sendToInflux.sh: line 9: --data-binary: command not found
Data couln't be send to Callback, retrying...
BLE packet: [mac] 00 1110161a18a4c1385e4dd8007729500b74c5 -58
Temperature:  11.9
Humidity:  41
Battery voltage: 2.932 V
RSSI: -58 dBm

HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: 5939fdcd-43ac-11eb-81e1-0242ac120002
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.8.3
X-Request-Id: 5939fdcd-43ac-11eb-81e1-0242ac120002
Date: Mon, 21 Dec 2020 16:48:27 GMT

BLE packet: [mac] 00 1110161a18a4c138b2b99900c5275c0bde12 -84
Temperature:  19.7
Humidity:  39
Battery voltage: 3.038 V
RSSI: -84 dBm
```
sendToInflux.sh:
curl -i -u "user:pass" -XPOST http://12.34.56.78:8086/write?db=db0&precision=s --data-binary "myHouse,sensorname=$2 temperature=$3,calibratedHumidity=$6,voltage=$5 $7"

"Ping" Influx with Curl:

```
pi@pi:~/MiTemperature2 $ curl -sL -I http://12.34.56.78:8086/ping
HTTP/1.1 204 No Content
Content-Type: application/json
Request-Id: 1e0469cd-433b-11eb-8016-0242ac120002
X-Influxdb-Build: OSS
X-Influxdb-Version: 1.8.3
X-Request-Id: 1e0469cd-433b-11eb-8016-0242ac120002
Date: Mon, 21 Dec 2020 03:17:55 GMT
```

JsBergbau, 2020-12-21:
Hi Luzr

> curl -i -u "user:pass" -XPOST http://12.34.56.78:8086/write?db=db0&precision=s --data-binary "myHouse,sensorname=$2 temperature=$3,calibratedHumidity=$6,voltage=$5 $7"

You have to escape the ampersand `&` like `\&` like in the original sendToInflux.sh file in the repository.

Luzr, 2020-12-21:
JsBergbau,

Thanks for the help. That was it!

closed 2020-12-21 by Luzr