JsBergbau/MiTemperature2 #56: Docker build fails

dbluxo, 2020-12-21:
Hi, first of all thank you for your great project! 💯

I would like to use your software on my Raspberry Pi in Docker, unfortunately the Docker build fails with:

```
Sending build context to Docker daemon  316.9kB
Step 1/9 : FROM xavierrrr/xrrzero:stretchpython3.7
stretchpython3.7: Pulling from xavierrrr/xrrzero
31994f9482cd: Pull complete 
0e62b5dbedee: Pull complete 
2f4de8133c30: Pull complete 
9d4db4b0b711: Pull complete 
82cb2844df84: Pull complete 
615de40008cf: Pull complete 
2f64e792ad4a: Pull complete 
4171acf2106d: Pull complete 
922866fe3ee0: Pull complete 
a306030b8f1c: Pull complete 
38e601cf0581: Pull complete 
Digest: sha256:52d4a50a2a63fbfb23587fa5302cfa8209adda6ae64f7f88b971eca4cd46481c
Status: Downloaded newer image for xavierrrr/xrrzero:stretchpython3.7
 ---> bfefa81a0fdd
Step 2/9 : ENV LANG C.UTF-8
 ---> Running in a89ec52d01c7
Removing intermediate container a89ec52d01c7
 ---> 0b0d69b1835e
Step 3/9 : RUN sudo apt-get -y --no-install-recommends install python-pip libglib2.0-dev
 ---> Running in a793593988a7
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libglib2.0-bin libglib2.0-data libpcre16-3 libpcre3-dev libpcre32-3
  libpcrecpp0v5 libpython-stdlib libpython2.7-minimal libpython2.7-stdlib
  python python-minimal python-pip-whl python2.7 python2.7-minimal
Suggested packages:
  libglib2.0-doc python-doc python-tk python2.7-doc binfmt-support
Recommended packages:
  python-all-dev python-setuptools python-wheel
The following NEW packages will be installed:
  libglib2.0-bin libglib2.0-data libglib2.0-dev libpcre16-3 libpcre3-dev
  libpcre32-3 libpcrecpp0v5 libpython-stdlib libpython2.7-minimal
  libpython2.7-stdlib python python-minimal python-pip python-pip-whl
  python2.7 python2.7-minimal
0 upgraded, 16 newly installed, 0 to remove and 33 not upgraded.
Need to get 13.6 MB of archives.
After this operation, 41.0 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian stretch/main armhf python-pip-whl all 9.0.1-2+rpt2 [1399 kB]
Get:2 http://archive.raspbian.org/raspbian stretch/main armhf libpcrecpp0v5 armhf 2:8.39-3 [149 kB]
Err:3 http://archive.raspbian.org/raspbian stretch/main armhf libpython2.7-minimal armhf 2.7.13-2+deb9u3
  404  Not Found
Err:4 http://archive.raspbian.org/raspbian stretch/main armhf python2.7-minimal armhf 2.7.13-2+deb9u3
  404  Not Found
Get:5 http://archive.raspbian.org/raspbian stretch/main armhf python-minimal armhf 2.7.13-2 [40.5 kB]
Err:6 http://archive.raspbian.org/raspbian stretch/main armhf libpython2.7-stdlib armhf 2.7.13-2+deb9u3
  404  Not Found
Err:7 http://archive.raspbian.org/raspbian stretch/main armhf python2.7 armhf 2.7.13-2+deb9u3
  404  Not Found
Get:8 http://archive.raspbian.org/raspbian stretch/main armhf libpython-stdlib armhf 2.7.13-2 [20.0 kB]
Get:9 http://archive.raspbian.org/raspbian stretch/main armhf python armhf 2.7.13-2 [154 kB]
Err:10 http://archive.raspbian.org/raspbian stretch/main armhf libglib2.0-data all 2.50.3-2+deb9u1
  404  Not Found
Err:11 http://archive.raspbian.org/raspbian stretch/main armhf libglib2.0-bin armhf 2.50.3-2+deb9u1
  404  Not Found
Get:12 http://archive.raspbian.org/raspbian stretch/main armhf libpcre16-3 armhf 2:8.39-3 [235 kB]
Get:13 http://archive.raspbian.org/raspbian stretch/main armhf libpcre32-3 armhf 2:8.39-3 [227 kB]
Get:14 http://archive.raspberrypi.org/debian stretch/main armhf python-pip all 9.0.1-2+rpt2 [179 kB]
Get:15 http://archive.raspbian.org/raspbian stretch/main armhf libpcre3-dev armhf 2:8.39-3 [565 kB]
Err:16 http://archive.raspbian.org/raspbian stretch/main armhf libglib2.0-dev armhf 2.50.3-2+deb9u1
  404  Not Found
Fetched 2969 kB in 0s (5248 kB/s)
E: Failed to fetch http://archive.raspbian.org/raspbian/pool/main/p/python2.7/libpython2.7-minimal_2.7.13-2+deb9u3_armhf.deb  404  Not Found
E: Failed to fetch http://archive.raspbian.org/raspbian/pool/main/p/python2.7/python2.7-minimal_2.7.13-2+deb9u3_armhf.deb  404  Not Found
E: Failed to fetch http://archive.raspbian.org/raspbian/pool/main/p/python2.7/libpython2.7-stdlib_2.7.13-2+deb9u3_armhf.deb  404  Not Found
E: Failed to fetch http://archive.raspbian.org/raspbian/pool/main/p/python2.7/python2.7_2.7.13-2+deb9u3_armhf.deb  404  Not Found
E: Failed to fetch http://archive.raspbian.org/raspbian/pool/main/g/glib2.0/libglib2.0-data_2.50.3-2+deb9u1_all.deb  404  Not Found
E: Failed to fetch http://archive.raspbian.org/raspbian/pool/main/g/glib2.0/libglib2.0-bin_2.50.3-2+deb9u1_armhf.deb  404  Not Found
E: Failed to fetch http://archive.raspbian.org/raspbian/pool/main/g/glib2.0/libglib2.0-dev_2.50.3-2+deb9u1_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
The command '/bin/sh -c sudo apt-get -y --no-install-recommends install python-pip libglib2.0-dev' returned a non-zero code: 100
```

Furthermore, I wonder what is behind `xavierrrr/xrrzero:stretchpython3.7`, I have not been able to find a source for it.

JsBergbau, 2020-12-21:
Hi dbluxo,

sorry I can't help you with docker. Hope some of the docker maintainers will help you soon. In the meantime I suggest using a Python venv for installing the dependencies.

mvadu, 2021-03-08:
Not to plug, but if you want to use the custom firmware versions (atc or pvvx) you can use [my docker image](https://hub.docker.com/repository/docker/adystech/mi_atc_reader). I went with config based approach to translate the mac to a name. I have not implemented all the nice features provided by this superb code though. I push it only to Influx, and working on MQTT as next target option.

CptK1ng, 2021-06-10:
To avoid the errors described by @dbluxo , we need just add `RUN sudo apt-get update` before the apt-get installations. The Dockerfile would look like this:
```
# Start from a Python3.7 base image I published

FROM xavierrrr/xrrzero:stretchpython3.7

# http://bugs.python.org/issue19846
# > At the moment, setting "LANG=C" on a Linux system *fundamentally breaks Python 3*, and that's not OK.
ENV LANG C.UTF-8

RUN sudo apt-get update

RUN sudo apt-get -y --no-install-recommends install python3-pip libglib2.0-dev

RUN /usr/local/bin/pip3.7 install bluepy
RUN /usr/local/bin/pip3.7 install requests

COPY LYWSD03MMC.py LYWSD03MMC.py
COPY sendtovera.py sendtovera.py

ENTRYPOINT service dbus start

#Change to the following line to match your needs

ENTRYPOINT while true; do /usr/local/bin/python3.7m LYWSD03MMC.py -d 44:C1:38:4B:A9:2A -r -b 100 --skipidentical 50 -deb --callback sendtovera.py; sleep 2; done

# start the created image with sudo docker run --net=host -t your_image_name /bin/bash
```

PR #103 (2021-11-27): Update Docker file to apt update
Fixes #56
commit ee9eec67 (2021-11-27): Update Docker file to apt update (#103)

Fixes #56
closed 2021-11-27 by JsBergbau