JsBergbau/MiTemperature2 #58: AttributeError: module 'socket' has no attribute 'AF_BLUETOOTH'

rjblake, 2020-12-31:
I've just installed and it runs normally, unless I select ATC mode (

```
pi@raspi:~/MiTemperature2 $ python3.9 ./LYWSD03MMC.py --atc --watchdogtimer 5
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a namefile and --namefileonlydevices is specified.
Also --name Argument is ignored, if you require names, please use --namefile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights, 
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Traceback (most recent call last):
  File "/home/pi/MiTemperature2/./LYWSD03MMC.py", line 423, in <module>
    toggle_device(dev_id, True)
  File "/home/pi/MiTemperature2/bluetooth_utils.py", line 113, in toggle_device
    hci_sock = socket.socket(socket.AF_BLUETOOTH,
AttributeError: module 'socket' has no attribute 'AF_BLUETOOTH'

```
I'm a bit lost on this and appreciate any suggestions. Some additional info as follows:

```
pi@raspi:~/MiTemperature2 $ sudo hcidump
HCI sniffer - Bluetooth packet analyzer ver 5.43
device: hci0 snap_len: 1500 filter: 0xffffffff
```
```
pi@raspi:~/MiTemperature2 $ sudo hcitool lescan
LE Scan ...
5F:D1:A9:B0:3D:9F (unknown)
E0:01:85:47:A6:53 (unknown)
E0:01:85:47:A6:53 Hue Lamp
A4:C1:38:9C:53:CF (unknown)
A4:C1:38:9C:53:CF LYWSD03MMC
A4:C1:38:4F:B1:F6 (unknown)
A4:C1:38:4F:B1:F6 LYWSD03MMC
F9:59:29:A0:E1:C4 (unknown)
68:B2:43:9C:47:1C (unknown)
68:B2:43:9C:47:1C (unknown
A4:C1:38:BC:6E:55 (unknown)
A4:C1:38:BC:6E:55 ATC_BC6E55
A4:C1:38:EF:E5:1A (unknown)
A4:C1:38:EF:E5:1A ATC_EFE51A

```

JsBergbau, 2020-12-31:
Hi @rjblake 
do you try to run Python 3.9 in a virtual environment? Do you have it completely installed? I think I had similiar errors when trying to run it from an additionally compiled version. Installing it completely and then installing 
`pip3 install bluepy pybluez` fixed it.
Without completly installing it, I was not able to install these 2 Python packages without any error.

rjblake, 2020-12-31:
I installed Python 3.9 on the Raspberry Pi (using instructions to build here [https://gist.github.com/SeppPenner/6a5a30ebc8f79936fa136c524417761d](url). i.e. built from source
I also installed bluepy and pybluez doing the following:
`pi@raspi:~/MiTemperature2 $ sudo pip3.9 install bluepy pybluez
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: bluepy in /usr/local/lib/python3.9/site-packages (1.3.0)
Requirement already satisfied: pybluez in /usr/local/lib/python3.9/site-packages (0.23)` and it reports as installed. When you say 'installing it completely', any suggestions

JsBergbau, 2020-12-31:
I don't know what exactly is the problem with Python, but I had the same problem. As soon as normal python is newer version it works.
Instead of `sudo make altinstall` do try `sudo make install`
I run on Python 3.9.1 compiled from source and it worked as soon it was installed over existing Python version.

rjblake, 2021-01-01:
@JsBergbau - thanks for the tip. Doing an `install` instead of `altinstall` appears to have fixed this. Wish there were cleaner ways of managing Python version as I've now I've got v2.7.x, v3.4.4, v3.5.3, v3.6.12, v3.9.1 - and still sometimes it'll pick 3.5.3 as default python3 despite having used `sudo update-alternatives --config python`  to use v3.9.1 - guess I've missed something somewhere

closed 2021-01-01 by rjblake

cloventt, 2022-07-07:
Sorry to necro this, but for future travellers you need to have Python built with Bluetooth socket support. This should happen automatically if you make sure that you have `bluetooth/bluetooth.h` available _before_ you build/install Python. It may not be available in your installation's default Python if Python was installed before the bluetooth lib. This can be installed like so:

```
sudo apt-get install libbluetooth-dev

# OR

sudo yum install bluez-libs-devel
```

Once the bluetooth libs are installed, then reinstall Python.

RAHEYO, 2024-04-04:
@cloventt what is the solution for macos system? apt-get and yum doesn't work on mac. I also cannot find brew nor macport packages named "libbluetooth-dev".

ktotam1, 2024-04-16:
same question but for windows, what's the workaround here? Also, it seems like bluepy is linux-only.