JsBergbau/MiTemperature2 #59: ATC compatible with Callback (domoticz)?

poseidogithub, 2021-01-03:
Hello,

thanks a lot for the effort on this. I'm giving it a try again as the non-ATC version never worked fine with me, sooner or later the process ate 100% of the processor and stoipped working.

Now I'm trying to use the ATC option, I was able to flash my sensors (big shoutout to @atc1441 in https://github.com/atc1441/ATC_MiThermometer) and I'm able to read my 6 sensors. with this line:

`./LYWSD03MMC.py  -a --watchdogtimer 5 --devicelistfile sensors.ini --onlydevicelist --callback sendToDomoticz.py`
In domoticz I have created dummy sensors (TEMP+HUMIDITY) for each one of my thermometers, I copied their IDX.

in my sensors.ini I have the devices, and their domoticz IDX:

```
[info]
info1=MAC Adresses must be in UPPERCASE otherwise sensor won't be found by the script
info2=now all available options are listet. If offset1, offset2, calpoint1 and calpoint2 are given 2Point calibration is used instead of humidityOffset.
info3= Note options are case sensitive
;info4=Use semicolon to comment out lines
sensorname=Specify an easy readable name
humidityOffset=-5
offset1 = -10
offset2 = 10
calpoint1 = 33
calpoint2 = 75

;[AA:BB:CC:DD:EE:FF]
; sensorname=Bathroom
; humidityOffset=-30
; offset1 = -10
; offset2 = -10
; calpoint1 = 33
; calpoint2 = 75

[A4:C1:38:6C:13:E1]
sensorname=86
[A4:C1:38:8D:46:23]
sensorname=87
[A4:C1:38:E9:0B:0E]
sensorname=88
[A4:C1:38:A6:2D:67]
sensorname=89
[A4:C1:38:9F:47:24]
sensorname=90
[A4:C1:38:C0:F3:DA]
sensorname=91
```
but in domoticz nothing appears. I wonder if sendtodomoticz.py has been updated to use ATC method...

I know that @ JsBergbau  is not a domoticz user, so maybe he could not help, anyone else can guide me?

JsBergbau, 2021-01-03:
I don't know what should be send to domoticz. If battery voltage, then change 
`val_bat = sys.argv[6]`
to
`val_bat = sys.argv[5]`
or if battery in percent should be reported, then try
`./LYWSD03MMC.py -a --watchdogtimer 5 -b --devicelistfile sensors.ini --onlydevicelist --callback sendToDomoticz.py` and it should work in current unmodified version.

poseidogithub, 2021-01-03:
> I don't know what should be send to domoticz. If battery voltage, then change
> `val_bat = sys.argv[6]`
> to
> `val_bat = sys.argv[5]`
> or if battery in percent should be reported, then try [...]

Thanks for the quick reply.

I think my "engrish" was not enough to explain my issue. sorry.

the thing is that your script works wonders reading my 6 sensors with the ATC firmware, this is completly fine and Im very grateful for your wok on this.

but my problem is that I dont see how sendToDomoticz.py can handle 6 sensors, as sendToDomoticz.py code identify only one IDX (sensor numers in domoticz) in line 10: idx_temp = "52", so from my newbie point of view, anything transferred trough callback to sendToDomoticz.py will be posted in domoticz on sensor IDX 52, because it's hardcoded and not a variable. 
having 6 xiaomi thermohygrometers means that in domoticz you need to have 6 sensors created wach one with a different IDX.

JsBergbau, 2021-01-03:
Now I understand. I think its not your english, but the fact, that I don't use domoticz, however please try changing
`idx_temp = "52"` 
to `idx_temp = sys.argv[2]`
then it should work in your case. Excellent work that you used the idx as sensorname in the configuration file.

poseidogithub, 2021-01-03:
> 
> 
> Now I understand. I think its not your english, but the fact, that I don't use domoticz, however please try changing
> `idx_temp = "52"`
> to `idx_temp = sys.argv[2]`
> then it should work in your case. Excellent work that you used the idx as sensorname in the configuration file.

that did the trick! fantastic!
thanks a lot

(maybe you should add some info on readme or sendToDomoticz.py for others to avoid having same issue?)

closed 2021-01-03 by poseidogithub

JsBergbau, 2021-01-03:
I've added a note and link to this issue in the sendToDomoticz.py file

GeitjeG, 2021-02-10:
I configured my sensor and I can get readings using `./LYWSD03MMC.py --atc`
However I cannot get it to send to Domoticz. I think because I use authentication. If I execute `./LYWSD03MMC.py -a --watchdogtimer 5 --devicelistfile sensors.ini --onlydevicelist --callback sendToDomoticz.py`, I get the following error:

```
During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/domoticz/plugins/MiTemperature2/sendToDomoticz.py", line 35, in <module>
    + val_bat
  File "/usr/local/lib/python3.7/dist-packages/requests/api.py", line 76, in get
    return request('get', url, params=params, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/requests/sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.7/dist-packages/requests/sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.1.52', port=80): Max retries exceeded with url: /json.htm?type=command&param=udevice&idx=943&nvalue=0&svalue=19.1;57;1&battery=1612953105 (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x75b76170>: Failed to establish a new connection: [Errno 111] Connection refused'))
Data couln't be send to Callback, retrying...
^CDisable LE scan
pi@raspberrypi:~/domoticz/plugins/MiTemperature2 $

```