JsBergbau/MiTemperature2 #60: Support BLE Advertisements with the official firmware?

matiaskorhonen, 2021-01-14:
The encryption scheme for the official firmware has been reverse engineered so it's possible to read the values from the BLE advertisement packets without having to flash the sensor with the custom ATC firmware.

Using the official firmware has the advantage that you don't need to recalibrate the temperature or humidity values…

For example:

* Python implementation: https://github.com/custom-components/ble_monitor/blob/ad07274a29278767ec4f2702cd28d1455eb90778/custom_components/ble_monitor/__init__.py
* Node.js implementation: https://github.com/hannseman/homebridge-mi-hygrothermograph/blob/2c959baa8df201d868c154da7a5de211edb1f716/lib/parser.js#L155-L197

This requires getting the encryption key (also known as the bind key) from the sensor, but this can be accomplished in using [the same page that's used to flash the ATC firmware](https://atc1441.github.io/TelinkFlasher.html).

JsBergbau, 2021-01-14:
Hi matiaskorhonen,

thanks for your request.

Just a few thougts

> Using the official firmware has the advantage that you don't need to recalibrate the temperature or humidity values…

I've noticed that with ATC firmware humidity value is more accurate than with original firmware. 

> Python implementation: https://github.com/custom-components/ble_monitor/blob/ad07274a29278767ec4f2702cd28d1455eb90778/custom_components/ble_monitor/__init__.py

This code looks good. Seems easy to implement and it is under MIT License, so I could adopt it.

I'll consider implementing it. 

Can you tell me how often original firmware sends encrypted data? A long interval held me back so far trying to go in that direction.

matiaskorhonen, 2021-01-14:
The README in the https://github.com/custom-components/ble_monitor repo says every 10 minutes for the LYWSD03MMC.

JsBergbau, 2021-01-14:
Sorry I was too fast in replying. Yes found also 10 minutes interval.

So developing and testing means waiting a lot.

And for me temperature values only every 10 minutes is a lot of time. And you aren't allowed to miss that data. If missed because of bad reception you have no updated data for 10 minutes.

So for me that original interval is far too long. For example I use these sensors detect if some forgot to close window in winter, when temperature drops too fast. Not possible with original advertisment data.

To get the bind key you have to use a software like the Telink flasher. So you have to connect to every device. When doing so you just could also flash the ATC firmware giving you much more possibilities. In fact using ATC firmware than original firmware could be even faster because you don't have to note the bind key and then set it up in the script. 

I can't say no to implementing it, however currently I have so many other fascinating projects and regarding the low benefits I can't priorize it at the moment.

If somebody is willing to help implementing it however I would do it. 

What I need for that:

* Byte sequence that reliably identifies an advertisment packet of original firmware with temperature and humidity data and battery data.
* python function that takes as parametes that advertisment packet as byte array and bind key. It then decodes what is decodable, so temperature and humidity and battery. These three values should be in a variable. 

The integration part in the script would I do.

JsBergbau, 2021-10-12:
Now custom encryted format is supported which sends data also every few seconds like the ATC version.