JsBergbau/MiTemperature2 #61: Script crashes in ATC mode

rsiska, 2021-01-16:
Script works fine (running with parameters: python3 LYWSD03MMC.py --atc --battery --callback sendToMQTT.sh), but eventually it crashes with following error:

Traceback (most recent call last):
  File "LYWSD03MMC.py", line 506, in <module>
    parse_le_advertising_events(sock, handler=le_advertise_packet_handler, debug=False)
  File "/home/pi/BLE/MiTemperature2-master/bluetooth_utils.py", line 345, in parse_le_advertising_events
    adv_type = struct.unpack("b", pkt[1:2])[0]
struct.error: unpack requires a buffer of 1 bytes

and after CTRL+C:

^CDisable LE scan
Exception ignored in: <module 'threading' from '/usr/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 1281, in _shutdown
    t.join()
  File "/usr/lib/python3.7/threading.py", line 1032, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.7/threading.py", line 1048, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
  File "LYWSD03MMC.py", line 42, in signal_handler
    disable_le_scan(sock)
  File "/home/pi/BLE/MiTemperature2-master/bluetooth_utils.py", line 230, in disable_le_scan
    bluez.hci_send_cmd(sock, OGF_LE_CTL, OCF_LE_SET_SCAN_ENABLE, cmd_pkt)
_bluetooth.error: (77, 'File descriptor in bad state')

JsBergbau, 2021-01-16:
Hallo rsiska,

looks like a very strange error. Error occurs in third party library for bluetooth.
How long runs the script fine bevor crashing? 

Can you try it running with `python3 LYWSD03MMC.py --atc --battery --watchdogtimer 5  --callback sendToMQTT.sh`

```
bluez.hci_send_cmd(sock, OGF_LE_CTL, OCF_LE_SET_SCAN_ENABLE, cmd_pkt)
_bluetooth.error: (77, 'File descriptor in bad state')
```
Looks like some internal blueooth error. What devices do use?

rsiska, 2021-01-16:
Hi, thank you for reply.
It always take not less than couple of hours to crash.
I am running it on Raspberry Pi 2, with bcm20702a0 Bluetooth adapter. (I know, not ideal.)
Running it with watchdog now.

JsBergbau, 2021-01-16:
Found this topic https://bugs.launchpad.net/ubuntu/+source/bluez/+bug/1886714 
There seems to exist a lot of problems with bcm20702a0 

> There are several reports of various equipments that disconnect, and I wonder if this could be the same problem.

> It may take 5–10 minutes and up to several hours before it disconnects.

Just two quotes from that bugreport. The text is quite long. If problem persists with watchdog I suggest using another bluetooth dongle.

rsiska, 2021-01-20:
It is running stable for 2 days now with watchdog. If that going to happen again, I will change the adapter.
Thank you! 
_Btw, I know it is not right place to do that, but what will be the right way to run this script as service with logging capabilities please?_

JsBergbau, 2021-01-20:
You could take a look here at zigbee2mqtt https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html#5-optional-running-as-a-daemon-with-systemctl how they run it as service.
Personally I start the script after reboot via cron an the `@reboot` command.

repohub, 2021-01-29:
I had the same issue, with Raspberry Pi 3 Model B Rev 1.2. Further I found a lot of bluetooth errors in my syslog and found this  bug at https://github.com/raspberrypi/firmware/issues/1150 with a suggested solution. The solution there did not help me put there seems to be a problem with the rpi-firmware. The problem now escaped after updating to the latest rpi-firmware and kernel.
Some more investigation is needed, but for now no bluetooth errors in the syslog and working ATC mode.

repohub, 2021-01-31:
From my point this issue could be closed. Syslog is still clean and LYWSD03MMC.py works without watchdog. The problem was the rpi-firmware on Raspberry Pi.

closed 2021-04-25 by JsBergbau