JsBergbau/MiTemperature2 #62: --count N does not supported with ATC ?

mkontta, 2021-01-21:
Hi,

I have case where:
- atc (1 device)
- count 2
- sendToDomoticz.py
- Raspi 3b+

Problem is that script does not exit after 2 sending. Any idea what is wrong?
Or do I have undertood "--count" option wrongly?

Used command and log below.

BR, Markku

pi@raspberrypi:~/MiTemperature2 $ ./LYWSD03MMC.py -a --watchdogtimer 5 --devicelistfile sensors.ini --onlydevicelist --callback sendToDomoticz.py -c 2 -b

Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a namefile and --namefileonlydevices is specified.
Also --name Argument is ignored, if you require names, please use --namefile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights,
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
BLE packet: A4:C1:38:2A:9E:26 00 1110161a18a4c1382a9e2600e018460b1ba2 -72
Temperature:  22.4
Humidity:  24
Battery voltage: 2.843 V
RSSI: -72 dBm
Battery: 70 %

/home/pi/MiTemperature2/sendToDomoticz.py sensorname,temperature,humidity,voltage,batteryLevel,timestamp 391 22.4 24 2.843 70 1611225510

BLE packet: A4:C1:38:2A:9E:26 00 1110161a18a4c1382a9e2600e018460b1ba3 -68
Temperature:  22.4
Humidity:  24
Battery voltage: 2.843 V
RSSI: -68 dBm
Battery: 70 %

/home/pi/MiTemperature2/sendToDomoticz.py sensorname,temperature,humidity,voltage,batteryLevel,timestamp 391 22.4 24 2.843 70 1611225532

BLE packet: A4:C1:38:2A:9E:26 00 1110161a18a4c1382a9e2600e018460b1ba4 -53
Temperature:  22.4
Humidity:  24
Battery voltage: 2.843 V
RSSI: -53 dBm
Battery: 70 %

/home/pi/MiTemperature2/sendToDomoticz.py sensorname,temperature,humidity,voltage,batteryLevel,timestamp 391 22.4 24 2.843 70 1611225585

JsBergbau, 2021-01-21:
Hi mkontta,

sorry count isn't supported right with atc-mode. Count was introduced to disconnect again from the sensor to save batterie. In ATC mode battery usage is the same, whether the script is running or not.

I see you use a list of sensors you want to monitor/report to callback. So how should `--count 2` behave here?
Exit when it has collected 2 measurements of your specified sensors? Exit when it has collected at least 2 measurements for each sensor? Exit when it has first collected 2 measurements of one of the sensors?

Since ATC mode receives multiple sensors at the same time and there is no disadvantage in keeping the script running and it is hard to decide how a count option should behave there was from my side of view no need for a count option. 

If you want to use it send multiple sensors to domoticz, please also have a look at there https://github.com/JsBergbau/MiTemperature2/issues/59

mkontta, 2021-01-22:
Hi,

Thank you for your answer. I understand what you mean.

In my case count could be e.g. 8 for 4 sensor. It is not so important if some sensor can not send data to Domoticz during period.

My idea was that script would execute with cron with selected interval and after that script could exit.

BR, Markku

JsBergbau, 2021-01-22:
How often would you run that script via cron?

mkontta, 2021-01-22:
Interval would be like 10min, 30min or 1 hour.

JsBergbau, 2021-01-22:
10 minute interval could be easily achieved when you set advertising interval to 10 minutes in Telinkflasher. Script only reports data to callback when ATC firmware has updated it (done via package counter). And then you have exactly that value. For your intended use case 8 packets for 4 sensor would mean you have to wait two advertising periods of the sensors and with that you have two reported values to domoticz.

If you still need longer intervals you could create a bash script which runs the script and sends a kill command lets say after 30 seconds. I would assist you if needed.

mkontta, 2021-01-23:
I will test this TelinkFlasher 10min option. Thanks for your support.

rjblake, 2021-02-01:
@mkontta - use caution setting the advertising interval too high in Telinkflasher - you will find it difficult to reconnect to the device if you want to change parameters in the future. The max is 10000msec from memory. To reset this, you'll need to use a tool like nRF Connect to reset (see [here](https://github.com/pvvx/ATC_MiThermometer/issues/25))

JsBergbau, 2021-02-01:
This is another advertising interval. Despite setting advertising interval to 10 seconds, still about every seconds is a BLE packet send. 
This setting concerns the interval advertising packets are updated, so every 10 seconds a new value or every minute and so on. With original ATC firmware this is no problem at all. Maybe in the fork of  pvvx for ATC firmware something changed. However I can confirm with original ATC firmware is no problem at all when changing advertising interval. I set all my thermoeters to 10 seconds and can connect flawlessly.

accountnumber3, 2022-05-05:
Here you go. Probably not perfect but it seems to be working for me the two times I tested it.

```py
554 elif args.passive:
[snip help and imports]
572         cnt=0 ####################
573         advCounter=dict()
574         #encryptedPacketStore=dict()
575         sensors = dict()
576         if args.devicelistfile:
```
```py
718                 def le_advertise_packet_handler(mac, adv_type, data, rssi):
719                         global cnt ###################
720                         global lastBLEPacketReceived
```
```py
751                                 print ("RSSI:", rssi, "dBm")
752                                 print ("Battery:", measurement.battery,"%")
753                                 print ("Count:", cnt)
754
##################
755                                 if args.count is not None and cnt >= args.count:
756                                         print(str(args.count) + " measurements colected. Exiting in a moment.")
757                                         os._exit(0)
758                                 cnt += 1
##################
759
760                                 currentMQTTTopic = MQTTTopic
761                                 if mac in sensors:
```