JsBergbau/MiTemperature2 #65: Sensors being read 5 times a second despite all being set to 10 min advertising interval

blairwigley, 2021-01-29:
For some reason my 4 sensors are being read every 4 seconds. 

This is what I'm calling:

`/home/pi/MiTemperature2/LYWSD03MMC.py --atc --battery --callback callback.py`

And callback.py just sends data to a Laravel setup:

`#!/usr/bin/python3 -u
#!/home/openhabian/Python3/Python-3.7.4/python -u
import requests

import sys

print(str(sys.argv[0]))

url = 'http://192.168.0.100:8000/temperature'
myobj = {
	'sensor': sys.argv[2],
	'temperature': sys.argv[3],
	'humidity': sys.argv[4],
	'battery': sys.argv[6],

}

x = requests.post(url, data = myobj)`

This is what's going to my db:

[image]

Any thoughts? Maybe I'm not using the Telink Flasher properly? I'm connecting to the already flashed devices and clicking "10 Minutes" under advertising interval. 

The Telink Flasher indicates that it's working:

`22:34:12: Connecting to: ATC_5835DB
22:34:17: Detected custom Firmware
22:34:25: Settings FE3C was send successful`

JsBergbau, 2021-01-29:
Hi,

looks very strange, especially because the sensor reports about every 10 seconds. However this should mean script works correctly, because data is transmitted via beacon every second no matter which interval you set. Setting interval causes the packet counter to change with the selected interval. So it really seems your sensor is set to a 10 seconds interval. 

`ATC_5835DB` this is another sensor then shown in screenshot. In screenshot is A1:0B:22

I'm not quite sure, but if I remember correctly sensor forgets settings when battery was removed. Perhaps this happened? 
Script reports every sensor. To use only specified sensors please have a look at 
https://github.com/JsBergbau/MiTemperature2#atc-mode-usage commands `--devicelistfile <filename>` and `--onlydevicelist`

You can also try setting another interval of 1 minute for example and check there. It may be possible that there is a bug in Telink Flasher or the firmware, don't know because I change interval to 10 seconds. However as far as I remember default setting of 1 minute worked, because I've forgotten to change with some sensors.

blairwigley, 2021-01-29:
Worked it out. 

Must have been something weird with the firmware I was using - it was receiving commands fine but not changing when I changed the smiley face or from C to F. 

So I reflashed all the the units and now it's working as expected. 

Thanks for your help!

closed 2021-01-29 by blairwigley