JsBergbau/MiTemperature2 #68: Only first sample in influxdb if LYWSD03MMC.py runs from shell?

bitbooster, 2021-02-07:
Hi,
I'm trying to replicate your example NodeRed flow, but with one difference: I have `./LYWSD03MMC.py --atc --watchdogtimer 5 --callback sendToNodeRed.sh` running from shell.
Then in my flow I have NodeRed waiting for the POST events, which are sent to influxdb, and then displayed via Grafana.

The issue I have is that if I run the script like that, Grafana shows only the first sample. 
To get influx to capture another sample I need to stop and restart the script.
While debugging the process, I could see the script sending more POSTs for all the samples sent by the sensor, NodeRed generating corresponding messages to the influxdb, and from the http log it seems also that influxdb does receive them as well.
Any idea then what could be wrong?

JsBergbau, 2021-02-07:
Hi bitbooster. Please mark your Node-RED flow, press CTRL + E and then copy the code and post it here.

bitbooster, 2021-02-08:
Hello, the flow is below here.

`[{"id":"bcff60bf.a6833","type":"tab","label":"MiTemperature2_flow","disabled":false,"info":""},{"id":"1b3ac735.d47e41","type":"http in","z":"bcff60bf.a6833","name":"","url":"/MiTemperature2Input","method":"post","upload":false,"swaggerDoc":"","x":140,"y":440,"wires":[["1be7e1cb.3ac0ee","cf0f669.1d5b418"]]},{"id":"1be7e1cb.3ac0ee","type":"change","z":"bcff60bf.a6833","name":"sensorlist","rules":[{"t":"set","p":"sensors","pt":"msg","to":"{\"info\":{\"info1\":\"MAC Adresses must be in UPPERCASE otherwise sensor won't be found by the script\",\"info2\":\"now all available options are listet. If offset1, offset2, calpoint1 and calpoint2 are given 2Point calibration is used instead of humidityOffset.\",\"info3\":\"Note options are case sensitive\",\"sensorname\":\"Specify an easy readable name\",\"humidityoffset\":-5,\"offset1\":-10,\"offset2\":10,\"calpoint1\":33,\"calpoint2\":75},\"A4:C1:38:73:DB:33\":{\"sensorname\":\"Mijia_01\",\"offset1\":-2,\"offset2\":2,\"calpoint1\":33,\"calpoint2\":75}}","tot":"json"},{"t":"set","p":"processOnlyListedSensors","pt":"msg","to":"true","tot":"bool"}],"action":"","property":"","from":"","to":"","reg":false,"x":400,"y":460,"wires":[["971d42fe.a2a9b","c0c5394d.88c4c8"]]},{"id":"971d42fe.a2a9b","type":"function","z":"bcff60bf.a6833","name":"Process and calibrate sensors","func":"processOnlyListedSensors = msg.processOnlyListedSensors || false;\n\nfunction calibrateHumidity2Points(humidity, offset1, offset2, calpoint1, calpoint2)\n{\n    p1y=calpoint1\n\tp2y=calpoint2\n\tp1x=p1y - offset1\n\tp2x=p2y - offset2\n\tm = (p1y - p2y) * 1.0 / (p1x - p2x)\n\tb = p2y - m * p2x //would be more efficient to do this calculations only once\n\thumidityCalibrated=m*humidity + b\n\tif (humidityCalibrated > 100 ) //with correct calibration this should not happen\n\t{\n\t    humidityCalibrated = 100\n\t}\n\telse if (humidityCalibrated < 0)\n\t{\n\t\thumidityCalibrated = 0\n\t}\n\thumidityCalibrated=Math.round(humidityCalibrated,0)\n\treturn humidityCalibrated\n}\n\n//\"A4:C1:38:69:E2:7C\"\nmsg.topic = msg.payload.sensor;\n\nif (msg.payload.sensor in msg.sensors )\n{\n    sensor =  msg.payload.sensor;\n    sensordaten = msg.sensors[sensor];\n    msg.payload.sensor = sensordaten.sensorname;\n    msg.topic = msg.payload.sensor;\n    if (\"offset1\" in sensordaten && \"offset2\" in sensordaten && \"calpoint1\" in sensordaten && \"calpoint2\" in sensordaten)\n    {\n        msg.payload.humidity = calibrateHumidity2Points(msg.payload.humidity, sensordaten.offset1, sensordaten.offset2, sensordaten.calpoint1,sensordaten.calpoint2);\n    }\n    else if (\"humidityOffset\" in sensordaten)\n    {\n        msg.payload.humidity = msg.payload.humidity + sensordaten.humidityOffset\n    }\n    return msg;\n}\nelse if (! processOnlyListedSensors)\n{\n    return msg;\n}\n//drop message because only listed sensors should be processed here","outputs":1,"noerr":0,"initialize":"","finalize":"","x":590,"y":520,"wires":[["94ff9630.20a2f8"]]},{"id":"94ff9630.20a2f8","type":"change","z":"bcff60bf.a6833","name":"Process for influxdb","rules":[{"t":"set","p":"measurement","pt":"msg","to":"tempMeasurement","tot":"str"},{"t":"set","p":"precision","pt":"msg","to":"s","tot":"str"},{"t":"set","p":"payload","pt":"msg","to":"[\t{\t    \"Temperature\" : $.payload.temperature,\t    \"humidity\" : $.payload.humidity,\t    \"voltage\": $.payload.voltage,\t    \"time\" : $floor($.payload.timestamp/25)*25\t},\t{\t    \"Sensorname\": $.topic\t}\t]","tot":"jsonata"}],"action":"","property":"","from":"","to":"","reg":false,"x":950,"y":520,"wires":[["613b21c6.d18e18","9e14f0.18f0c31"]]},{"id":"613b21c6.d18e18","type":"influxdb out","z":"bcff60bf.a6833","influxdb":"4fa2494f.6b8a48","name":"","measurement":"","precision":"","retentionPolicy":"","database":"","retentionPolicyV18Flux":"","org":"","bucket":"","x":1230,"y":580,"wires":[]},{"id":"cf0f669.1d5b418","type":"debug","z":"bcff60bf.a6833","name":"Sensor Raw","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":360,"y":640,"wires":[]},{"id":"c0c5394d.88c4c8","type":"debug","z":"bcff60bf.a6833","name":"OutFromSensorlist","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":610,"y":460,"wires":[]},{"id":"9e14f0.18f0c31","type":"debug","z":"bcff60bf.a6833","name":"GoToInfluxdb","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":1220,"y":520,"wires":[]},{"id":"4fa2494f.6b8a48","type":"influxdb","z":"","hostname":"influxdb","port":"8086","protocol":"http","database":"temp_mijia","name":"","usetls":false,"tls":"1846b94d.d950e7","influxdbVersion":"1.x","url":"","rejectUnauthorized":false},{"id":"1846b94d.d950e7","type":"tls-config","name":"","cert":"","key":"","ca":"","certname":"m2mqtt_srv.crt","keyname":"m2mqtt_srv.key","caname":"m2mqtt_ca.crt","verifyservercert":false}]`

JsBergbau, 2021-02-08:
You've changed the sample flow and deleted the http Node
[image]
Without that it can't work. If you add it, like in the sample, it works.

closed 2021-02-08 by JsBergbau