JsBergbau/MiTemperature2 #71: Issues with connection

skibbipl, 2021-02-17:
I've switched from RPi4 (where script worked) to RPi3 and I'm trying to connect my ATC flashed sensors to it. Unfortunately I'm having the same issue as in #14. However I can connect to the sensor with `gatttool`:
```
gatttool -I
[                 ][LE]> connect XX:XX:XX:XX:XX:XX
Attempting to connect to XX:XX:XX:XX:XX:XX
Connection successful
Notification handle = 0x0012 value: e5 00
Notification handle = 0x0015 value: f0 0a
Notification handle = 0x000e value: 2b
Notification handle = 0x0012 value: e5 00
Notification handle = 0x0015 value: f0 0a
Notification handle = 0x000e value: 2b
Notification handle = 0x0012 value: e5 00
Notification handle = 0x0015 value: f0 0a
Notification handle = 0x000e value: 2b
```
Any ideas what might be wrong?

JsBergbau, 2021-02-17:
Hallo skibbipl,

What commandline do you use to read the values of your sensors?

With ATC sensors you shouldn't connect at all, but just run the script with --atc in ATC mode.

skibbipl, 2021-02-17:
`./LYWSD03MMC.py -d XX:XX:XX:XX:XX:XX -a`

closed 2021-02-17 by skibbipl