JsBergbau/MiTemperature2 #72: Wrong values/frames

chrisschn, 2021-02-23:
Hi, 1st of all: I love this :+1:  - so easy getting the sensors connected to my homeassistant setup.

But from time to time I have obv. wrong values - messing my graphs :cry: . I tracked them in the logs:

```
Feb 22 20:15:33 raspberrypi run-xiaomi.sh[14362]: BLE packet: A4:C1:38:75:02:12 00 1110161a18a4c1387502120052000b7e1db5 4
Feb 22 20:15:33 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
Feb 22 20:15:33 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
Feb 22 20:15:33 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
Feb 22 20:15:33 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
Feb 22 20:15:33 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)
Feb 22 20:15:33 raspberrypi run-xiaomi.sh[14362]: Temperature:  8.2
Feb 22 20:15:33 raspberrypi run-xiaomi.sh[14362]: Humidity:  0
Feb 22 20:15:33 raspberrypi run-xiaomi.sh[14362]: Battery voltage: 32.285 V
Feb 22 20:15:33 raspberrypi run-xiaomi.sh[14362]: RSSI: 4 dBm
Feb 22 20:15:33 raspberrypi run-xiaomi.sh[14362]: Battery: 11 %
Feb 22 20:15:34 raspberrypi run-xiaomi.sh[14362]: /home/pi/MiTemperature2/sendToMQTT_HA.sh sensorname,temperature,humidity,voltage,batteryLevel,timestamp SensorX 8.2 0 32.285 11 1614024933

```

This is how it looks like normally:
```
Feb 23 14:04:00 raspberrypi run-xiaomi.sh[14362]: BLE packet: A4:C1:38:75:02:12 00 1110161a18a4c13875021200e728520b7ca4 -78
Feb 23 14:04:00 raspberrypi run-xiaomi.sh[14362]: Temperature:  23.1
Feb 23 14:04:00 raspberrypi run-xiaomi.sh[14362]: Humidity:  40
Feb 23 14:04:00 raspberrypi run-xiaomi.sh[14362]: Battery voltage: 2.94 V
Feb 23 14:04:00 raspberrypi run-xiaomi.sh[14362]: RSSI: -78 dBm
Feb 23 14:04:00 raspberrypi run-xiaomi.sh[14362]: Battery: 82 %
```

My setup in short:
 - raspberry pi 2, raspbian buster
 - script running as systemd service calling:  `$XIAOMI_HOME/LYWSD03MMC.py --atc -b 1000 -r --debounce --skipidentical 50 -df $XIAOMI_HOME/sensors.ini --callback sendToMQTT_HA.sh`

Gonna try to sort these values out with the python-script and avoid publishing them via MQTT, but for now I do not know how.
Maybe someone has an idea?

JsBergbau, 2021-02-23:
Hi chrisschn
looks like a quite complex problem, especially `Feb 22 20:15:33 raspberrypi kernel: Bluetooth: hci0: Frame reassembly failed (-84)` 
For me it looks like a hardware/firmware/driver issue. Take a look at there https://github.com/raspberrypi/firmware/issues/1150 hope that helps you.

chrisschn, 2021-02-26:
THX. Updating raspi firmware seems to make it - so far.

closed 2021-02-26 by chrisschn