JsBergbau/MiTemperature2 #73: RSSI not being read

g0lgs, 2021-02-28:
I am not getting RSSI values reported by the script (if I use a callback I can see it is always sending a 0 values)

```
python3 ./LYWSD03MMC.py -d A4:C1:38:BD:FC:24 --battery --round --debounce --rssi 
Trying to connect to A4:C1:38:BD:FC:24
Temperature: 25.5
Humidity: 39
Battery voltage: 3.022 V
Battery level: 92

Temperature: 25.5
Humidity: 39
Battery voltage: 3.022 V
Battery level: 92

```

JsBergbau, 2021-02-28:
Hi g9lgs,

RSSI is only available in ATC Mode, thats why this command is in the section "ATC related arguments". 
I found no way to get RSSI in connected mode. According to https://stackoverflow.com/a/24417605/11890443 for Bluetooth LE devices the only way to get RSSI is while scanning. ATC mode scans all the time, thats why we get RSSI there.

g0lgs, 2021-02-28:
Thanks for the Clarification - I had noticed that --devicelistfile required ATC mode but missed that --rssi requiredfthat too.

closed 2021-02-28 by g0lgs