JsBergbau/MiTemperature2 #75: more sensors reading

elanozturk, 2021-03-15:
hello, i have two sensors and i would like to get data from them in one terminal. I tried to add my sensors mac address to sensors.ini file but didn't work. I also created a file which contains 2 mac address with --onlydevicelist option but no luck. Can you please explain how to do that?

JsBergbau, 2021-03-15:
Hi elanozturk,
please post your full commandline and your full sensors.ini file here.

elanozturk, 2021-03-15:
tried `./LYWSD03MMC.py --devicelistfile list.txt `(contains mac address), `./LYWSD03MMC.py -odl`

sensors.ini:

[info]
info1=MAC Adresses must be in UPPERCASE otherwise sensor won't be found by the script
info2=now all available options are listet. If offset1, offset2, calpoint1 and calpoint2 are given 2Point calibration is used instead of humidityOffset.
info3= Note options are case sensitive
;info4=Use semicolon to comment out lines
sensorname=Specify an easy readable name
humidityOffset=-5
offset1 = -10
offset2 = 10
calpoint1 = 33
calpoint2 = 75

;[A4:C1:38:A1:DB:88]
; sensorname=Outdoor
; humidityOffset=-30
; offset1 = -10
; offset2 = -10
; calpoint1 = 33
; calpoint2 = 75

[A4:C1:38:B1:05:06]
sensorname=Indoor
offset1 = -2
offset2 = 2
calpoint1 = 33
calpoint2 = 75

JsBergbau, 2021-03-15:
Thanks. Now I see.
Devicelistfile only works for ATC mode
```
ATC mode related arguments:
  --atc, -a             Read the data of devices with custom ATC firmware
                        flashed, use --battery to get battery level
                        additionaly in percent
  --watchdogtimer X, -wdt X
                        Re-enable scanning after not receiving any BLE packet
                        after X seconds
  --devicelistfile DEVICELISTFILE, -df DEVICELISTFILE
                        Specify a device list file giving further details to
                        devices
  --onlydevicelist, -odl
                        Only read devices which are in the device list file
  --rssi, -rs           Report RSSI via callback
```

elanozturk, 2021-03-15:
thank you, how should i do this in atc mode?

JsBergbau, 2021-03-15:
Take a look at https://github.com/atc1441/ATC_MiThermometer how to get them into ATC mode.
I suggest then Flashing software from here https://github.com/pvvx/ATC_MiThermometer This is the most advanced software currently available which saves most battery.

elanozturk, 2021-03-15:
Thank you, appreciated for your help.

elanozturk, 2021-03-15:
One last question, if i flash the custom firmware am i able to use mi home app for check the temp history? and can i re-install stock firmware after custom?

JsBergbau, 2021-03-15:
You can reinstall the original firmware. However I don't think that the will still work with mi home app. You can share your results here.