JsBergbau/MiTemperature2 #77: atc mode readings

xjustbmw, 2021-03-18:
For some reason my bluetooth doesnt connect to sensors if i specify each mach with -d option

in atc mode it works. reads them all 
I can specify to read only one sensor, but not only one time
-c 1 doesn't seem to work on -atc mode

any suggestions ?

xjustbmw, 2021-03-18:
# ./LYWSD03MMC.py -d  A4:C1:38:42:40:6A -a
Trying to connect to A4:C1:38:42:40:6A
Connection lost
Waiting...
Trying to connect to A4:C1:38:42:40:6A
Connection lost
Waiting...
Trying to connect to A4:C1:38:42:40:6A
^CConnection lost

# ./LYWSD03MMC.py -a
....
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
BLE packet: A4:C1:38:42:40:6A 00 1110161a18a4c13842406a00e929530b8605 -80
Temperature:  23.3
Humidity:  41
Battery voltage: 2.95 V
RSSI: -80 dBm

BLE packet: A4:C1:38:77:1E:F5 00 1110161a18a4c138771ef500f7254f0b6326 -76
Temperature:  24.7
Humidity:  37
Battery voltage: 2.915 V
RSSI: -76 dBm

BLE packet: A4:C1:38:8A:68:8D 00 1110161a18a4c1388a688d00de2a520b7a2c -91
Temperature:  22.2
Humidity:  42
Battery voltage: 2.938 V
RSSI: -91 dBm

# ./LYWSD03MMC.py -a -odl --devicelistfile sensors.ini -c 1
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a devicelistfile and --onlydevicelist is specified.
Also --name Argument is ignored, if you require names, please use --devicelistfile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights,
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
BLE packet: A4:C1:38:42:40:6A 00 1110161a18a4c13842406a00e929530b890d -91
Temperature:  23.3
Humidity:  41
Battery voltage: 2.953 V
RSSI: -91 dBm
Humidity calibrated (2 points calibration):  40

.. doesnt stop after one reading

JsBergbau, 2021-03-18:
Hi xjustbmw
`--count` is not supported, with more info about that see https://github.com/JsBergbau/MiTemperature2/issues/62

xjustbmw, 2021-03-18:
right on. 
the main problem is that some of the sensors are not received all the time, since the signal is not so great for all of them, setting advertising too high would be hit and miss for the constant temperature readings.

so instead of moving the receiver in the middle, where i have no power supply available,
i will bash it out to send the last read temperature every 10-15 minutes for each sensor, should be more than enough :) 

not so sure why it wont connect directly to the sensor, but for battery sake maybe its better this way ?

many thanks for your contribution

JsBergbau, 2021-03-18:
Have you already tried this https://github.com/pvvx/ATC_MiThermometer alternative ATC firmware? With that I have a longe range. than with "original" ATC firmware.
With ATC mode you can have multiple receivers. But perhaps you don

xjustbmw, 2021-03-18:
yes. this is the firmware that i am using on all my sensors, its set at maximum transmission power at the moment
i was using mijiatemp android to connect to them then today I bought an asus 400 bluetooth dongle for raspberrypi
i dont know the bluetooth reception quality of the pi zerow, but for me the wifi connection is not stable enough, even near the wifi router

JsBergbau, 2021-03-18:
Very interessting because I have also very stable Wifi with Pi Zero W.
Tipp: Set Advertisting mode to "ATC1441" of thermostat instead of all. This doubles or triples the chance of receiving the advertisment.
If you have really problems with wifi, I would connect a Wifi dongle to the PI Zero W (or an usb LAN Adapter) and use the integrated bluetooth module. My external bluetooth dongle was much worse then the integrated of PI Zero W.

xjustbmw, 2021-03-18:
probably a defective PI unit. there are many people online with unstable wifi on pi zeroW. something strangely also related to the HDMI connection/resolution.

# cat data.txt|awk '{print $2}'|sort|uniq -c
    345 A4:C1:38:42:40:6A
     74 A4:C1:38:75:E8:21
    446 A4:C1:38:77:1E:F5
    337 A4:C1:38:8A:68:8D

sensor data received at the moment in the same time interval :) only 74 packets from the sensor that is far away :) still very usable . im not interested to see the same temperature at each second ;) 
 
will test with advertising ATC1441 mode only

xjustbmw, 2021-03-18:
went 'the hardware' way. replaced the sensor position :) works well now

xjustbmw, 2021-03-25:
any way to limit the output from this --atc mode to like one reading every ~5 minutes per sensor ?

JsBergbau, 2021-03-25:
You can change the interval of the sensor to 5 minutes then script only outputs every 5 minutes.

xjustbmw, 2021-03-25:
then it is still possible to connect to the sensor ?

JsBergbau, 2021-03-25:
Not connecting but callback reports back every 5 minutes. Does that fit your requirements?

xjustbmw, 2021-03-25:
i now have it at 2500ms
that is the advertising interval default right for pvvx firmware ?

i remember that i read somewhere that if you set it higher than 10000ms it will fail to connect to the sensor after that

JsBergbau, 2021-03-25:
[image]
You set the Measure interval. So if you have 2500 ms advertising interval and want a new measurement every 5 minutes you calculate 5 * 60 / 2,5 = 120
So your new measure interval is 120. Please try.

xjustbmw, 2021-03-25:
Its not the measure that is a problem, but the advertising interval.

Im using a bash/curl  script from LYWSD03MMC.py callback function to write to the database. and that will get huge at 2.5 seconds rate and 4 sensors.
https://github.com/JsBergbau/MiTemperature2/blob/master/sendToInflux.sh

I think this was discussed already,the right way would be to average these readings before writing to the database every 5 minutes

Before, I did this filtering through node-red, but the ble scanner crashes after a while, and that one does not have the watchdog function enabled (like the LYWSD03MMC python).

But, LYWSD03MMC.py has a function to skip identical measurements to be sent to the callback script, If i add the measure interval "120" and set that parameter high, it sould work ;)

xjustbmw, 2021-03-25:
hmm.. aparently the maximum value for the measure interval is 25

JsBergbau, 2021-03-25:
> and that will get huge at 2.5 seconds rate and 4 sensors.

The rate is 10 seconds at default, because callback is only executed when adv packet number changes and this changes every 4 * 2500 ms.

If you are concerned about your database size there is a much better option. Change sendToInflux to:

```
timestamp=$7
interval=300
timestamp=$(expr $timestamp / $interval \* $interval + $interval)

curl -i -u "user:pass" -XPOST http://<host>/write?db=openhab_db\&precision=s --data-binary "AquaraBluetoothSensors,sensorname=$2 temperature=$3,calibratedHumidity=$6,voltage=$5 $7"
```
That stores always exakt at 5 minutes. And after passing such a 5 minute chunk you have the latest value bound to that 5 minute chunk. First value after that is stored in a few minutes in the future, but than as time goes by it is correct again. Sorry hard to explain. And because the values are stored at always the same interval influx uses RLE encoding. Instead of like 600-900 Byte for the timestamps it only uses about 9 bytes. That saves really much space.

xjustbmw, 2021-03-26:
So at the moment I have it at 25x5000ms (125 seconds measurements)
Database size seems to be quite low even with measurements set at 10 seconds interval.
How do you store your sensors data?

There is also precision=m and retention policy can be applied to it so it doesnt get huge in few ..years :)
Also the very old data can be averaged each hour in another history database.

many thanks for your help

JsBergbau, 2021-03-26:
> Database size seems to be quite low even with measurements set at 10 seconds interval.
Indeed it is. Influxb uses special algorithms to keep a lot of data with few space.

> How do you store your sensors data?
Also with influxdb. Sensors send new data every 10 seconds and with the formula above I store one value each 25 seconds. There is no averaged data. I keep all original data.

xjustbmw, 2021-03-26:
all clear
resource wise. if the same ble scanner/influx database is running on localhost
its faster to write the data using influx command ?

like this (from documentation)
influx write \
  -b bucketName \
  -o orgName \
  -p s \
  'myMeasurement,host=myHost testField="testData" 1556896326'

JsBergbau, 2021-03-26:
Can't tell that. Try it with prepending time command, do a lot of writes and average the time command.
However this measures only the time consumed, not how much resources. It would take some time to explain and in this case these ressources used here aren't worth to spend too much time on it.