JsBergbau/MiTemperature2 #78: Support for pvvx/ATC_MiThermometer firmware

luckfamousa, 2021-04-07:
I have flashed some devices with the alternative firmware from https://github.com/pvvx/ATC_MiThermometer which has some more options than https://github.com/atc1441/ATC_MiThermometer but I cannot receive any events from these devices while events from devices flashed with https://github.com/atc1441/ATC_MiThermometer are happily received. 

Any chance to support https://github.com/pvvx/ATC_MiThermometer as well?

JsBergbau, 2021-04-07:
Please post your commandline and a screenshot of your configuration of your sensor of https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

I use pvvx version on all my sensors and it works flawlessly, so we'll also get yours to work.

luckfamousa, 2021-04-07:
That's great!

I call it currently like this 

```bash
python3 ./LYWSD03MMC.py --atc 
```

And then I see this:
```
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a devicelistfile and --onlydevicelist is specified.
Also --name Argument is ignored, if you require names, please use --devicelistfile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights, 
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
BLE packet: A4:C1:38:8D:92:A6 00 1110161a18a4c1388d92a600e523550ba126 -72
Temperature:  22.9
Humidity:  35
Battery voltage: 2.977 V
RSSI: -72 dBm

BLE packet: A4:C1:38:8D:92:A6 00 1110161a18a4c1388d92a600e523550ba127 -72
Temperature:  22.9
Humidity:  35
Battery voltage: 2.977 V
RSSI: -72 dBm
```

Where A4:C1:38:8D:92:A6 is the device that was flashed with atc1441/ATC_MiThermometer.

Here is a screenshot (one random device):
[Telink Flasher v2.9c.pdf](https://github.com/JsBergbau/MiTemperature2/files/6272831/Telink.Flasher.v2.9c.pdf)

JsBergbau, 2021-04-07:
Looks ok so far.
Please try `python3 ./LYWSD03MMC.py --atc --watchdogtimer 5` and try bring the flashed sensor closer to your receiver.

luckfamousa, 2021-04-07:
Now the sensor is placed only 2 cm away from the receiver (Raspberry Pi 4) and I used your command but still just getting events from A4:C1:38:8D:92:A6.

JsBergbau, 2021-04-07:
Can you flash another sensor? Perhaps this one is faulty in some way.

JsBergbau, 2021-04-07:
Here are the settings of one of my sensors. 
You could try increaseing the TX power. This shouldn't matter but it is worth a try.
[image]

fz6, 2021-04-07:
Hi,

I was experiencing a similiar issue. I think it's a problem with v2.9 of pvvx's firmware (change-log says `Added additional flags to advertising packages` so it would make sense). I flashed v2.7 instead and now see updates:

```
python3 LYWSD03MMC.py --atc --watchdogtimer 5
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a devicelistfile and --onlydevicelist is specified.
Also --name Argument is ignored, if you require names, please use --devicelistfile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights, 
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
BLE packet: A4:C1:38:3E:90:3A 00 1110161a18a4c1383e903a00bd334a0b330d -61
Temperature:  18.9
Humidity:  51
Battery voltage: 2.867 V
RSSI: -61 dBm

^CDisable LE scan
```

Running v2.9 (tried advertising mode set to `all` and `Atc1441`) I would just see `Listening ...` until I gave up.

FWIW, I initially thought it may be a faulty sensor, or I'd done something wrong (first try using custom firmware), but I managed to to get output using package `aioblescan` (`python3 -m aioblescan -A`) with both firmware versions.

Personally I think the firmware should be fixed... stick to defined format for `Atc1441` and use a different UUID for the new custom format pvvx implemented.

Either way, happy my sensor is working again :grin:

JsBergbau, 2021-04-09:
The format of firmware Version 2.9 is now supported. Version 3 supports now both formats. To save a few percent battery please set it that way 
[image]

AdFlags checked: 35 Bytes Paket length, format of firmware version 2.9, device is shown on Android Bluetooth scan.

AdFlags not checked: 32 Bytes Paket length, format like prior versions, device is not shown on Android Bluetooth scan.

fz6, 2021-04-09:
Thanks for the update, I'm going to stick with the version I've flashed after reading the firmware authors rationale for the change (and the changelog for v3...). All working well here and managed to get readings from a sensor at the far end of my house, which saves me rigging up a second receiver :+1:

luckfamousa, 2021-04-15:
Sorry for the delay. I did not have the chance to test the new version earlier. 
However, after updating I do no more get any events. Neither from the sensor that used to work before nor the other sensors with the pvvx firmware.

JsBergbau, 2021-04-15:
Can you please fetch the lastest version, reboot whole system and try `./LYWSD03MMC.py --atc --watchdogtimer 5` and post your output?

luckfamousa, 2021-04-15:
Done.
Here is the output (one sensor again 2 cm away from the RPI).

```
:~/MiTemperature2 $ python3 ./LYWSD03MMC.py --atc --watchdogtimer 5
Script started in ATC Mode
----------------------------
In this mode all devices within reach are read out, unless a devicelistfile and --onlydevicelist is specified.
Also --name Argument is ignored, if you require names, please use --devicelistfile.
In this mode rounding and debouncing are not available, since ATC firmware sends out only one decimal place.
ATC mode usually requires root rights. If you want to use it with normal user rights, 
please execute "sudo setcap cap_net_raw,cap_net_admin+eip $(eval readlink -f `which python3`)"
You have to redo this step if you upgrade your python version.
----------------------------
Power ON bluetooth device 0
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
```

JsBergbau, 2021-04-15:
Hi luckfamousa,

thank you very much for testing. Please update, now it should work.

luckfamousa, 2021-04-15:
Getting flooded with events now :) Thank you for your great support here!

luckfamousa, 2021-04-17:
Everything works now.

closed 2021-04-17 by luckfamousa