JsBergbau/MiTemperature2 #79: When will the procedure for parsing Advertising packages be corrected in accordance with the Bluetooth LE standard?

pvvx, 2021-04-09:
When will the procedure for parsing Advertising packages be corrected in accordance with the Bluetooth LE standard?
An advertisement package has the right to contain several AD-sections. This is standard in BLE.
[image]

https://github.com/JsBergbau/MiTemperature2/blob/master/LYWSD03MMC.py#L445-L450 ?

JsBergbau, 2021-04-09:
It worked perfectly for original ATC1441 firmware advertisments :)
Then a new advertisment format came up which was only distinguishable from orginal ATC1441 format by its length. If not filtering this paket wrong measurements were decoded. Sometimes we code things and even have no clue that this code might break compatibility with other solutions.

However I'll do my best to improve it. Can you give me a hint how to easily change it to support v2.9 without breaking compatibility?

pvvx, 2021-04-09:
https://github.com/pvvx/ATC_MiThermometer/issues/79#issuecomment-808807418

This flags parameter gives the device visibility in standard applications for Windows, Android, ...
In the current firmware version 3.0, the addition of "Ad-flags" is switched in the settings and depends on the user's choice.

pvvx, 2021-04-09:
> Then a new advertisment format came up which was only distinguishable from orginal ATC1441 format by its length. 

This is enough to determine that the data structure is different.

> However I'll do my best to improve it. Can you give me a hint how to easily change it to support v2.9 without breaking compatibility?

Parse the ad package in the same way as all other software does.

Length PDU, AD length, AD Type, AD data, [ end Length PDU? ]  ADn length, ADn Type, ADn-data...

Known  AD-Type? id = 16-bit UUID? -> Known UUID? -> we process by AD-Length AD-data, ...

JsBergbau, 2021-04-09:
Support for Firmware v2.9 is now added. Thank you very much for also supporting the shorter "old" packet format with 32 Bytes length which is active when checkbox is not checked:
[image]

When checkbox is checked packet length is 35 bytes and is also now decoded correctly
[image]

closed 2021-04-25 by JsBergbau