JsBergbau/MiTemperature2 #80: add quiet / daemon mode

gazzat5, 2021-04-13:
Hi there,
firstly, I just wanted to say great work on this program! It's super useful.

Now, I was just wondering if you could please add a flag to supress console output, for example -q?
Only for when a callback is specified.
For running as a service for example.

I know i could just redirect output to /dev/null but it seems neater to do it at the program level.

Cheers

JsBergbau, 2021-04-13:
Hi gazzat5,

thanks for your appreciation. 

Adding a switch like `-q` is some work due to how the program works.
You could write a wrapper script that redirects output if it is important for you.

Also if you want to run it as service you can specify
`StandardOutput=null` in service unit file, see https://www.freedesktop.org/software/systemd/man/systemd.exec.html

gazzat5, 2021-04-14:
Wouldn't it be possible to change all print statements to point to a function that only prints if the quiet flag isnt set?

It would still be useful to have this feature as error messages could still be outputed which would go to syslog etc. 

Thanks for the tip on systems I'll try that for now

JsBergbau, 2021-04-25:
Yes it is possible. I prever a pragmatic solution. The goal you want to achive is currently possible without any effort. Rewriting the script takes time which can be used for other useful stuff like new features or other cool projects. Personally I prefer new features and other projects than doing the last 20 % on something, see https://en.wikipedia.org/wiki/Pareto_principle