JsBergbau/MiTemperature2 #81: sendToDomoticz.py errors

siklosi, 2021-04-13:
Hi,
there are few errors in sendToDomoticz.py
first of all
val_bat = sys.argv[6] 
sys.argv[6] is actualy timestamp and sys.argv[5] is voltage. 
Also domoticz accepts percentage (0-100) for battery 

I modified this file for my self like this:

def translate(value, leftMin, leftMax, rightMin, rightMax):
    leftSpan = leftMax - leftMin
    rightSpan = rightMax - rightMin
    valueScaled = float(value - leftMin) / float(leftSpan)
    return rightMin + (valueScaled * rightSpan)

val_bat = str(translate(float(sys.argv[5]),1.8,3.3,0,100))

I'm assuming that max is 3.3 (I just bought 9 of this sensors and some are reporting 3.23) and for low I set it at 1.8 but I really don't know what is lowest voltage on which this sensors work.

JsBergbau, 2021-04-13:
Hi siklosi,

arg number depends on how you call the script. Take a look there https://github.com/JsBergbau/MiTemperature2/issues/59#issuecomment-753654764

> I'm assuming that max is 3.3 (I just bought 9 of this sensors and some are reporting 3.23) and for low I set it at 1.8 but I really don't know what is lowest voltage on which this sensors work.

My sensors reportet at maximum 3,34V and lowest voltage reported so far was 0.44 V. This low voltage was probably measured while trasmitting data.

jenshenk, 2021-04-29:
Okay, after fiddling around a bit more I found the error. Maybe it's interesting for the next update: 
For some reason if I change the val_temp = sys.argv[5] to 5 (thought I go for the calibrated one) this error happens. If I set it back to the default value (so val_temp = sys.argv[3]) this issue does not happen.

mywildsoul, 2022-01-25:
After updating the LYWSD03MMC.py script two days ago, I get this error:

File "/home/pi/domoticz/scripts/python/MiTemperature2/LYWSD03MMC.py", line 666
    else
        ^
SyntaxError: invalid syntax

JsBergbau, 2022-01-25:
Hi mywildsoul,

thanks for reporting this. There was a missing colon from this pull request [112](https://github.com/JsBergbau/MiTemperature2/pull/112)
I've fixed it. Please update again and it should work now. Please give short feedback.

mywildsoul, 2022-01-25:
Thanks a lot!
Working perfect :)