JsBergbau/MiTemperature2 #83: sendTo script using Homie convention

gos-fr, 2021-04-29:
Hi !

I wrote a small script to send the data from your tool to a MQTT server using the Homie convention. I have it working with OpenHAB 3.
[sendToHomie.zip](https://github.com/JsBergbau/MiTemperature2/files/6400546/sendToHomie.zip)
Feel free to add it to your project.

And also thank you for this nice tool !

JsBergbau, 2021-04-29:
Hi gos-fr,

thanks for your contribution. Which device do you use as receiver and how many sensors do you receive?
I'm asking because executing 27 times mosquitto_pub for one sensor may overload a raspberry PI Zero W quite fast. If then network connection was gone it could take days to catch up measurements. 
Is it perhaps possible to send it in one JSON string? In a few hours or tomorrow I'll post an update with support for integrated MQTT broker, because via Callback it takes a lot of CPU usage and time with just only one callback executed.

gos-fr, 2021-04-29:
I'm running on a virtual machine on a Synology NAS, with only 3 sensors. So, no problems with performance. But you're right, that's a lot of calls.
Unfortunately the Homie convention does not support JSON.
But in fact, only the last 4 mosquitto_pub calls are needed to update the values. The calls before that are to setup the data structure and it would be enough to send them on startup. One could put them into a separate script that's called on system start, especially since Homie requires the MQTT messages to be retained. But I was too lazy to do that :D

The integrated MQTT broken is an interesting update. I'll be sure to look into that.

JsBergbau, 2021-05-01:
Hi @gos-fr 
Version 3 with integrated MQTT support is out.
Thanks for providing your script. It came to the right time. Without it Version 3 wouldn't have the subtopics option.