JsBergbau/MiTemperature2 #84: NameError: name 'topic' is not defined

mstaroste, 2021-05-03:
Hello,

I copied the sample mqtt.conf file from the repository to my working folder.
When starting I receive the following error

`Fehler
name 'topic' is not defined
Traceback (most recent call last):
  File "./LYWSD03MMC.py", line 252, in handleNotification
    myMQTTPublish(topic,jsonString)
NameError: name 'topic' is not defined
`

Can anyone help?

Thanks.

JsBergbau, 2021-05-03:
Hi mstaroste,

how exactly do you run the script?

mstaroste, 2021-05-03:
Hello JsBergbau,

`./LYWSD03MMC.py -d a4:c1:38:b1:54:d6 -r -b --mqttconfigfile mqtt.conf`

and the result is

---------------------------------------------
MiTemperature2 / ATC Thermometer version 3.0
---------------------------------------------
Trying to connect to a4:c1:38:b1:54:d6
MQTT connected with result code 0
Temperature: 23.6
Humidity: 47
Battery voltage: 3.059 V
Battery level: 96
Fehler
name 'topic' is not defined
Traceback (most recent call last):
  File "./LYWSD03MMC.py", line 252, in handleNotification
    myMQTTPublish(topic,jsonString)
NameError: name 'topic' is not defined

itpony, 2021-05-03:
I have the same issue. My quick fix was to change line 252 of LYWSD03MMC.py to this:

`myMQTTPublish(MQTTTopic,jsonString)`

I have no idea if it breaks anything, but it is publishing to MQTT now

JsBergbau, 2021-05-03:
Thanks for reporting this bug. It is now fixed. Check out v3.1
@itpony indeed that was the correct solution.

My recommendation: Try out ATC mode. Battery lasts much longer.