JsBergbau/MiTemperature2 #87: Node Red Callback and Dashboard

Sukalo0290, 2021-05-23:
Apologies, i'm pretty new to all this. I'm struggling with MQTT so using the Callback flow.

I want to display the gauges and charts directly on the Node Red Dashboard so i have two questions about this. firstly the device is connecting and bringing in the advertisements using the atc, the packet seems to be split up firstly which is proving difficult to use the split or function to extract the individual measurements to send the temperature to one graph and humidity to another.

[image]

Also where is this data or how to i setup the http as i don't seem to get any data from this area.
[image]

Thanks for any assistance.

JsBergbau, 2021-05-23:
Your data is recognised as string and not as JSON. 

What is your commandline to send the data to node red? Whats inside your callback script? Did you edit the IP address to point to your Node-RED instance?

Sukalo0290, 2021-05-23:
So i imported the Callback file which populates all the nodes for me, I then deploy and press the start button
[image]
then it produces the string as you say in the debug window.

I haven't done anything else.

JsBergbau, 2021-05-23:
You have to edit sendToNodeRed.sh because there is the configuration with IP address to Node-RED.

Sukalo0290, 2021-05-23:
Okay, what do i need to edit exactly?

JsBergbau, 2021-05-24:
This is sendToNodeRed.sh

Please edit at the places in angle brackets `< >`

```
#!/bin/bash

#use e.g with that script: MySensor.sh 
#!/bin/bash
#DIR="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null 2>&1 && pwd )"
#$DIR/LYWSD03MMC.py --atc --watchdogtimer 5 --callback sendToNodeRed.sh

if [[ "$1" == *"rssi"* ]]
then
	curl -s -i -XPOST http://<YOURHOST>:<YOURPORT>/MiTemperature2Input -H "Content-Type: application/json" --data "{\"sensor\": \"$2\", \"temperature\" : $3 , \"humidity\" : $4 , \"voltage\" :$5, \"rssi\" : $6, \"timestamp\": $7 }"
else
	curl -s -i -XPOST http://<YOURHOST>:<YOURPORT>/MiTemperature2Input -H "Content-Type: application/json" --data "{\"sensor\": \"$2\", \"temperature\" : $3 , \"humidity\" : $4 , \"voltage\" :$5, \"timestamp\": $6 }"
fi
```

Sukalo0290, 2021-05-24:
Thanks for the response, i have actually managed to get the MQTT working today so now i'm trying to connect the influxdb so i can show the results on grafana or something but there doesn't seem to be any input yet so still working on it. I'm sure i'll be back if can't sort it.

closed 2021-06-03 by Sukalo0290